istoreos/istoreos #15: 配置完旁路有 iStoreOS没有网 下面的lan口有网

wanglongfei8899, 2022-04-17:


jannson, 2022-04-17:
可能旁路由里面的 CPU 没配置 DNS，你手动配置一个 DNS 里面就有网络了。请试试。。。

wanglongfei8899, 2022-04-17:
![image](https://user-images.githubusercontent.com/75508757/163708415-23560c85-7866-4e0b-90bd-9e4bcfbd2c5

wanglongfei8899, 2022-04-17:
[image]
[image]

wanglongfei8899, 2022-04-17:
是不是我哪里设置有问题  但是可以正常上网 就是固件不能 更新软件包 也ping不通外网

wanglongfei8899, 2022-04-17:
主路由是开HDCP的   iStoreOS是关闭DHCP的

wanglongfei8899, 2022-04-18:
奇葩 给iStoreOS换个地址 iStoreOS有网了 然后下面LAN口没网了 还能这样

jannson, 2022-04-19:
核心就是 DNS 错误导致没网络。你把DNS 改成：223.5.5.5 223.6.6.6 试试。
可以在路由器里面 ping 223.5.5.5 试试

mmmqx827, 2022-04-20:
我的小主机软路由安装istoreOS后能够上网，但是安装网页提供的第三方科学项目的插件，配置好后显示未启用，重启也一样，现在passwall和ss都是这个情况，麻烦问一下怎么解决，谢谢

jannson, 2022-04-22:
旁路由问题解决了，我关闭这个 issue了。第三方的插件问题，会有第三方的人修复的。

closed 2022-04-22 by jannson