istoreos/istoreos #18: 【问题汇总】以及问题优先级

jannson, 2022-04-17:


closed 2022-04-18 by jannson

99477204, 2022-04-18:
希望软件商店增加迅雷快鸟

jannson, 2022-04-19:
迅雷快鸟 Merlin 我们曾经开发过，但是 OP 下没有。你有推荐的 ipk 么？有的化我们编译并放到软件中心

99477204, 2022-04-19:
我下载的这个https://github.com/sensec/luci-app-xlnetacc/releases/tag/v1.0.5.1，但是安装后并不运行。
我之前用的一个openwrt里边的快鸟插件非常稳定，登录后一直没有掉过线，如果可以希望你们研究下提取出来放在软件中心里。因为我从他们公布的插件包里下载的迅雷快鸟安装后，依然是不运行的。附之前用的固件地址，链接:https://www.123pan.com/s/dS5A-By4qd 提取码:SeY7，我之前用的是x86的 4月1号发布的plus大全版。
另外附上他们的GitHub地址https://github.com/sirpdboy/openwrt

lifaxian, 2022-04-19:
宝哥，samba不支持中文目录文件，下个版本有望修复吗？

jannson, 2022-04-21:
> 宝哥，samba不支持中文目录文件，下个版本有望修复吗？

那天我看到你反馈的问题了，然后我直接记录了，并且给我同事测试。他说没问题，中文时支持的。你方便可以加群里，或者截图给我看看。在你的 issue 里面截图，别在这里。会 miss 掉

jiaoxiaojia, 2022-04-22:
我的3865U只要挂载另外再挂载一个硬盘，系统在线更新就会报GTP的错误，把硬盘卸载在线更新就会正常，这是bug还是？

jannson, 2022-04-22:
> 我的3865U只要挂载另外再挂载一个硬盘，系统在线更新就会报GTP的错误，把硬盘卸载在线更新就会正常，这是bug还是？

这个问题新版本已经修复。

jiaoxiaojia, 2022-04-26:
用了几个版本的istoreOS，温度显示好像不管用，永远显示27度

wildtigerw, 2022-04-30:
想要jellyfin啊。。。。。。。。

genesis-lua, 2022-05-05:
> 想要jellyfin啊。。。。。。。。

jellyfin用docker安装没有吗

nemo830, 2022-05-09:
宝哥 发现机场节点无法代理ipv6的ip 访问油管没有问题 看奈飞就无法登录 把wan6删了重启就可以了 也不知道是机场的问题 还是固件的问题 删了ipv6 却影响PT上传 因为现在很多人都用ipv6下载了

sll2046, 2022-05-10:
j3455下exsi6.7安装，设置双线wan负载均衡网速不能叠加反而降速，同设置下原LEDE固件可以正常叠加。

jiaoxiaojia, 2022-05-18:
更新到5.18的固件后，固件一直重复提示有新固件，更新了几次还是一直提示

gintokighost, 2022-06-12:
istore插件卸载后重新安装这个插件的配置不能初始化，还是保留的删除以前的设置

gintokighost, 2022-06-12:
[插件卸载后，重新安装此插件配置并没有初始化，测试插件DDNS.TO，卸载重装后密钥还是以前的密钥，不能设置为空，为空就报错！！！！

HoyoEcho, 2022-06-14:
宝哥宝哥，可以加入Full-Cone NAT功能吗，家里想挂个网心云

ablvsqrn, 2022-06-29:
> > 宝哥，samba不支持中文目录文件，下个版本有望修复吗？
> 
> 那天我看到你反馈的问题了，然后我直接记录了，并且给我同事测试。他说没问题，中文时支持的。你方便可以加群里，或者截图给我看看。在你的 issue 里面截图，别在这里。会 miss 掉

mount.cifs //10.10.10.20/5e38/onecloud/  /mnt -o user=root,pass=123456,iocharset=utf8
我的也有着问题 但是我在网上找到上面的命令改成我的路径就可以解决了
我原来的命令是mount -t cifs -o username=root,password=123456,  //10.10.10.20/5e38/onecloud/  /mnt
我不是程序员来着，这个方法可以解决中文乱码

ChinaParrot, 2022-06-30:
x86 的root登陆页面密码怎么修改，命令修改不生效。

zzxysun, 2022-07-08:
大佬，能不能增加pptp vpn, 上网时间控制，app控制？

zzxysun, 2022-07-08:
大佬，r68s能不能增加pptp vpn, 上网时间控制，app控制？

jannson, 2022-07-08:
上网时间控制，app控制 我们考虑中。。。

VwEl, 2022-07-20:
R2S核心频率有上 1.6G嘛 可以搭配‘CPU 性能优化调节设置’和‘CPU占用率限制’

cqchuckie, 2022-08-08:
一直使用LEDE，iStoreOS什么时候添加softenter和unifi控制器？迫不及待想使用iStoreOS了

molin830, 2022-08-14:
> 迅雷快鸟 Merlin 我们曾经开发过，但是 OP 下没有。你有推荐的 ipk 么？有的化我们编译并放到软件中心 @[99477204](https://github.com/99477204)

大佬哥 好像不支持226的网卡呀

molin830, 2022-08-14:
小宝大佬哥 好像不支持226的网卡呀 刷完进不去后台

molin830, 2022-08-15:
大佬 能不能添加226网卡驱动呀

ChinaParrot, 2022-08-15:
ufi-003随身wifi能支持一下么?

molin830, 2022-08-21:
都没有人在x86 i226网卡 物理机下进不去后台 网口无法使用的问题吗

jefoky, 2022-09-17:
希望能支持D2550板子，刷U盘和ssd，D2550都无法启动

gzxiexl, 2022-10-07:
希望可以选择不带Docker的版本 因为我是玩aio，本身不需要在op里跑docker，istoreos因为带了docker ，内存我要给到1G才行，实测512M会因为爆内存卡死。

xcl0797, 2022-11-17:
提示no interfaces found
quikstart
0
打开直接提示这个，估计是网卡驱动未兼容，麻烦看看能不能增加一下，网卡i225b3，N5105的机子，畅网第三版豪华版！目前用的esir高大全版本，网卡是兼容在里面的，所以可能对你们来说或许没有很难

Q1364630478, 2022-12-05:
r66s无法挂载ntfs，格式化成ext4也无法挂载。用命令行修复显示找不到。另外可以出一个插件类似airconnect嘛，让dlna设备支持AirPlay

jian263994241, 2022-12-08:
希望增加 n5105 核显驱动 .

nevertoday, 2022-12-09:
另外可以出一个插件类似airconnect嘛，让dlna设备支持AirPlay
_---
同求，刚需！

sgzman, 2023-02-03:
快速配置aria2的时候，如果已挂载硬盘名称为sata1-1，并且选择了下载路径。然后服务-Aria2中下载目录为【/mnt/sda1/download】，需要手动更改路径为【/mnt/sata1-1/download】才能启动aria

zhengliuluhong, 2023-03-31:
系统->软件包，点击配置OPKG,报错，‘’无法读取 /etc/opkg.conf：Error: 未指定的错误‘’，现在是无法安装软件，麻烦看看是什么问题呢？我在ssh里面看了etc目录下的opkg.conf，文件不见了。我用winscp无法复制进去。

277354688, 2023-04-08:
希望iStore软件能随时更新，部分软件不是最新版本，尤其是Alist这种对版本敏感的软件，非常需要随时更新。

wizlau, 2023-04-19:
2023-04-07 x86版本测试发现重启后upup服务会失效，重启后upnp服务需要关闭再打开才能使用

PineApple5201314, 2023-04-22:
宝哥 为什么我在安装完homeassistant之后，命令行显示成功，但是页面状态是未运行？

hassanfei, 2023-07-05:
21.02.3 2023060914版本，使用系统设置里面的重启，总是死机，无法进入web后台界面，必须断电在送电操作才能进入web界面。长时间等待也依然不进系统。

gzxiexl, 2023-07-05:
> 21.02.3 2023060914版本，使用系统设置里面的重启，总是死机，无法进入web后台界面，必须断电在送电操作才能进入web界面。长时间等待也依然不进系统。

X86-efi 版本，我也是遇到了这个情况，翻墙软件无效了，web界面无法进入。

Huyang-boy, 2024-04-04:
> 21.02.3 2023060914版本，使用系统设置里面的重启，总是死机，无法进入web后台界面，必须断电在送电操作才能进入web界面。长时间等待也依然不进系统。

我也遇到了，怎么解决的？ssh可以连接，reboot命令无效

hzqdg, 2024-07-10:
闲置的玩客云做旁路由很合适，希望出个玩客云的旁路由专版！

myzhongqiang, 2025-07-05:
> 21.02.3 2023060914版本，使用系统设置里面的重启，总是死机，无法进入web后台界面，必须断电在送电操作才能进入web界面。长时间等待也依然不进系统。

iStoreOS 24.10.1, 2025060614 X86，我也遇到了怎么解决啊