istoreos/istoreos #21: 【开发计划】下一步的功能

jannson, 2022-04-17:
## 功能计划

- [ ] 首页的向导的每个功能，增加高级入口
- [ ] 沙箱模式，保证大家折腾失败可以直接退回老版本
- [ ] 在线升级，所有已安装的插件不丢失，配置也不丢失

cnfjice, 2022-04-17:
能否把端口改回来，eth0默认是lan，eth1是wan，其他端口为未分配，自行添加。。。

jannson, 2022-04-18:
目前默认 eth0 是 WAN，其它是 LAN，这样大家比较容易记住。高级用户可以自行去删除，再添加等等

langslove, 2022-04-18:
> 

eth0作为wan口这个设定我觉得是最好的，我也搞不懂恩山那些固件为什么都把ETH1做成wan口，太同质化了

langslove, 2022-04-18:
> 目前默认 eth0 是 WAN，其它是 LAN，这样大家比较容易记住。高级用户可以自行去删除，再添加等等

下个版本什么时候发布啊宝哥

jannson, 2022-04-19:
我估计周五。目前要修复我置顶的几个 BUG，特别是升级的 BUG，端口转发环回无效，还加上几个驱动的。

houyantao, 2022-04-19:
目前在格式虚拟机安装比较繁琐，是否可以考虑出一个VMDK和VHD格式的虚拟磁盘格式，或者做一个这类格式的简单的过渡固件。

loonyd, 2022-04-20:
i225驱动现在完善了么？小宝老大。刚入了n5105，打算换回koolshare了

jannson, 2022-04-20:
貌似支持了，请测试下。

oldboyooxx, 2022-04-22:
周五了，我来催你了，23333333
----
是不是直接【在线升级】就可以了！←这点我真的很喜欢

jannson, 2022-04-22:
对的，我们还紧张测试中。不过记得要备份插件，现在升级插件是会丢失的。

wssfxtx, 2022-04-22:
>我觉得，eth0是lan，ethn是wan，中间的默认都是lan

jannson, 2022-04-22:
之前想过这个方案，但是这样的话 wan 不是固定的，总觉得不够完美。目前固件也发布了，文档也写了，不打算修改了。

langslove, 2022-04-22:
> 

宝哥，周五了，等你的固件等的好辛苦啊

jannson, 2022-04-22:
[image]

jannson, 2022-04-22:
最新消息。开始编译了。。。但是我们的中文控制台，中文乱码，先不解决了。尴尬

中文控制台时支持在终端直接修改 IP等的功能。结果在显示器的终端中文乱码，尴尬

langslove, 2022-04-22:
> 

没事，看的懂英文的，先用呗

jannson, 2022-04-22:
更新啦，亲们可以测试了。

RichKyle, 2022-04-24:
求一个之前的家长控制插件

jannson, 2022-04-24:
> 求一个之前的家长控制插件

会考虑的，软件我们都开发好了：KoolProxy。但是需要很多规则的维护，这个是很需要精力的

houyantao, 2022-04-25:
希望能增加cloudflare的动态域名支持

genesis-lua, 2022-04-29:
> 希望能增加cloudflare的动态域名支持

这个是支持的，安装了ddns之后在原生软件包里面搜索cloudflare有一个插件，安装他，之后在DDNS里面就可以选到这个，我正在用

houyantao, 2022-04-29:
> > 希望能增加cloudflare的动态域名支持
> 
> 这个是支持的，安装了ddns之后在原生软件包里面搜索cloudflare有一个插件，安装他，之后在DDNS里面就可以选到这个，我正在用

谢谢

hpqqph, 2022-04-30:
能否加入bypass的支持

genesis-lua, 2022-04-30:
> 能否加入bypass的支持

附议，bypass好用

caesaranthony, 2022-05-09:
建议在iStore里增加Adguard home
curl -s -S -L https://raw.githubusercontent.com/AdguardTeam/AdGuardHome/master/scripts/install.sh | sh -s -- -v
用官方提供的脚本，也安装不上
谢谢

jannson, 2022-05-18:
在别人的离线安装包项目当中，有了。

caesaranthony, 2022-05-18:
> 在别人的离线安装包项目当中，有了。

请教，如何安装在iStore中，谢谢

jannson, 2022-05-19:
iStore 上传安装就行了。

L1cardo, 2022-06-02:
> > 
> 
> eth0作为wan口这个设定我觉得是最好的，我也搞不懂恩山那些固件为什么都把ETH1做成wan口，太同质化了

这不叫同质化，openwrt就是这么设置的。第一个添加的网口是lan，第二个是wan。

gintokighost, 2022-06-09:
ssr离线安装失败

yaoshuangshl, 2022-06-18:
Adguardhome会集成到istore商店中嘛~

byteinfo, 2022-06-27:
R4S 重启和reboot 不能正常启动，一直亮红灯，必须重插电源才能开机，几个版本都是这样。。。

chainone, 2022-07-09:
帮忙把wireguard的支持也加进下一步里：
https://github.com/linkease/istoreos/issues/290

byteinfo, 2022-07-18:
> R4S 重启和reboot 不能正常启动，一直亮红灯，必须重插电源才能开机，几个版本都是这样。。。
之前Balenaetcher写固件无法重启，这次用rufus写固件竟然可以重启了。。。

byteinfo, 2022-07-27:
R4S 上的USB插了移动硬盘， S.M.A.R.T. 创建计划任务没有硬盘可选

abcc666, 2022-08-05:
> 在别人的离线安装包项目当中，有了。

帮忙把kmod-nft-compat、kmod-ipt-socket支持也加进下一步里吧，可以参考这个报错信息，在istoreOS上用也跟截图固件的报错一样(https://github.com/kiddin9/openwrt-packages/issues/60#top)

hotwa, 2022-08-08:
添加一下tailscale吧
[tailscale](https://github.com/adyanth/openwrt-tailscale-enabler)

zhangtaidong, 2022-08-09:
softetherVPN 希望添加到iStore

Free189, 2022-08-12:
AdGuard Home安装后在服务里找不到，3000可以进网页

jannson, 2022-08-15:
这几个 VPN 的插件感觉问题不大。

Free189, 2022-08-15:
这是一封自动回复邮件。已经收到您的来信，我会尽快回复。

molin830, 2022-08-17:
小宝大神 什么时候加入支持i226网卡 物理机下网口不能用
5.15的内核可以 5.4的不行

lycleft1, 2022-08-20:
感觉用window 10 的Hyper-V操作安装部署最简单，但是没有对应的vhd 和vhdx对应文件导入，大神可否开发一个？

wssfxtx, 2022-10-11:
您的邮件已经收到，我将尽快回复，谢谢！

wuzj1121, 2022-10-13:
istoreos怎么安装AdGuardHome？我要疯了，怎么都安不上！！！！！！！！

Paretoa, 2022-10-17:
是否可以添加wireguard支持

2003jie, 2022-10-31:
什么时候能出r4se的固件

unpack8866, 2022-11-01:
能否把ipv6搞正常了，从爱快和op转过来的，发现ipv6不正常

jac88288, 2022-11-05:
能把MariaDB  php7.3 phpMyAdmin融合一起，那么更强大！这样可以建站服务

abcc666, 2022-11-11:
> kmod-nft-compat、kmod-ipt-socket

最新版已有kmod-nft-compat，希望宝哥在后续版本加入kmod-ipt-socket支持

erqing-lvxv, 2022-11-15:
我看到iStore里Jellyfin和plex等影音插件，有没有考虑过为iStoreOS添加英特尔核显驱动

GreenTeodoro839, 2022-11-15:
把螃蟹网卡的kmod集成进去吧

bingle456l, 2022-11-26:
大佬能否让应用商店的emby开启硬解，或者能否加入N5105的核显驱动，自己docker部署emby可以device /dev/dri:/dev/dri驱动核显

Saika50, 2022-11-26:
新发布的R5S固件刷进tf卡以后启动不了。。。直接进了emmc里的官方固件

krkrchan, 2022-11-27:
终于看到R2C了，宝哥威武！

jian263994241, 2022-12-08:
啥时候支持 n5105 啊, 希望增加核显驱动 .  我看到别的固件已经支持了, 馋呀

Ejoyyee, 2023-01-01:
期待看到R5C的固件~~~

wssfxtx, 2023-01-01:
您的邮件已经收到，我将尽快回复，谢谢！

PandaEgg, 2023-01-13:
可以增加PPTP 等vpn客户端和服务端吗

woann, 2023-03-31:
这个问题到现在也没有解决，就是文本框的内容修改了，提交会提示“没有待应用的更改”，就是压根没监听文本框的内容变更。其他的op固件不存在这个问题，就istoreos有这个问题

[4 more replies]