istoreos/istoreos #23: x86 刚刷完机后 检测升级 报错

zb737472783, 2022-04-18:
[image]

这个怎么解决啊...

zb737472783, 2022-04-18:
parted -l  用这个命令修复后 现在报错 
[image]

jannson, 2022-04-18:
下个版本修复，可以考虑先手动升级。当根目录有其它分区的时候，我们的升级判断有问题。

zb737472783, 2022-04-18:
收到!

gonboy, 2022-04-18:
原来2G调整为10G 后， 升级导致第2(10G)，3个(50G)分区内容 全没了。 现在后续不敢改2G的大小了。
还是用默认的2G吧

jannson, 2022-04-18:
分区不能随便改，2G  安装软件足够用了。如果实在不够用，以后也会有沙箱模式（多个 overrlay 分区）的扩容办法。总之最好别自己去调整

zzyhdn0928, 2022-04-19:
(https://fw.koolcenter.com/iStoreOS/x86_64/openwrt-21.02.1-2022041521-x86_64-squashfs-combined.img.gz)这个版本升级仍然报错 离线安装软件也报错

zzyhdn0928, 2022-04-19:
[image]

zb737472783, 2022-04-19:
> [image]

一样,等下个版本

zzyhdn0928, 2022-04-19:
[image]

zzyhdn0928, 2022-04-19:
[image]

jannson, 2022-04-19:
我知道了，很可能你路由器里面没网络。。。所以检测都失败了。你路由器里面，有个网络工具，ping www.baidu.com 试试。

zzyhdn0928, 2022-04-21:
[image]
[image]
ping 百度也不通

zzyhdn0928, 2022-04-21:
旁路由模式

jannson, 2022-04-21:
你的DNS 是自动获取的，你手动改 DNS：223.5.5.5
试试。应该就有网络了，然后就可以安装并且在线升级了

zzyhdn0928, 2022-04-21:
好的 我试试

jannson, 2022-04-22:
先手动升级，下个版本就可以了。修复了这个问题。

closed 2022-04-22 by jannson