istoreos/istoreos #33: 考虑支持中文控制台

jannson, 2022-04-19:
类似爱快这个，但是最好是简化版本，比如可以更改内网 IP 神马的：

[image]

lifaxian, 2022-04-19:
这个必须顶顶，让小白更容易上手！！！

TZFcdd, 2022-04-19:
这个真的可以有

Wwwu6, 2022-04-20:
> 这个真的可以有

复议

jannson, 2022-04-21:
已经搞了个简单的，这个周末会随着大版本发布。

jannson, 2022-04-22:
更新了一个简单的版本。不过在显示器上的交互模式还有问题，可以用单个命令模式。

jannson, 2022-05-18:
中文乱码，改成英文支持了。

```
root@iStoreOS:~# quickstart
✔ 0、Show Interfaces
Interfaces:
-----------
● LAN
  ● IPv4: 192.168.100.1
  ● IPv6:
  ● DeviceName: br-lan
  ● Devices:
    ● Name: eth1
    ● Link: DOWN
    ● Speed:
    ● Name: eth2
    ● Link: DOWN
    ● Speed:
    ● Name: eth3
    ● Link: DOWN
    ● Speed:
● WAN
  ● IPv4: 192.168.10.231
  ● IPv6:
  ● DeviceName: eth0
  ● Devices:
    ● Name: eth0
    ● Link: UP
    ● Speed: 1000 Mbit/s
● WAN6
  ● IPv4:
  ● IPv6:
  ● DeviceName: eth0
  ● Devices:
    ● Name: eth0
    ● Link: UP
    ● Speed: 1000 Mbit/s

Use the arrow keys to navigate: ↓ ↑ → ←
? Console:
  ▸ 0、Show Interfaces
    1、Change LAN IP
    2、Install X86
    3、Reset
    q、QUIT

```

jannson, 2022-05-18:
感觉没毛病，先关闭了。以后想增加功能再说。

closed 2022-05-18 by jannson