istoreos/istoreos #51: NFS也共享不成功。

lingyunzhiss, 2022-04-22:
[image]
[image]
[image]

jannson, 2022-04-23:
NFS 这个共享，之前我们测试发现，如果 NFS 连接断开了，就会造成假死状态。NFS 适合的网络是那种网络很稳定长期不掉线的。所以我们不常用。如果说有问题我们可以看看，优先级低一点。

你的另外一个帖子说的 Samba 是每期上线必须测试的。是没问题的。

lingyunzhiss, 2022-04-23:
> NFS 这个共享，之前我们测试发现，如果 NFS 连接断开了，就会造成假死状态。NFS 适合的网络是那种网络很稳定长期不掉线的。所以我们不常用。如果说有问题我们可以看看，优先级低一点。
> 
> 你的另外一个帖子说的 Samba 是每期上线必须测试的。是没问题的。

谢谢你回复我。Samba能用了，用\\10.0.10.1收到后在W11下固定到快速访问，大概3小时后网络计算机列表列出来了
[image]
NFS共享我发现成功了，同样是用\\Lede.lan的方法查找的，在KODI下能使用，飞快。就是在局域网无法显示名称。
另外：迫切需要多硬盘休眠呀，毕竟节省硬盘寿命些。谢谢。

lingyunzhiss, 2022-04-23:
硬盘休眠搞定，软件包里有。谢谢