istoreos/istoreos #55: LAN口的物理设置问题，请指教

horse02, 2022-04-23:
小宝大神，你好，我用虚拟机虚拟你们的固件，用下来赶紧很稳定，也能在线升级固件，非常方便，针对小白特别友好，感谢小宝大哥的辛苦付出，不过里面有个网络接口的LAN口设置有点不明白，是你们固件问题还是我小白不懂。LAN口里怎么没有物理设置呢？我软路由的LAN口有3个，其中一个已经ETH1设置好了，我怎么再设置ETH2,ETH3呢？在你们固件里找不到啊，请告知一下，谢谢啦

jannson, 2022-04-23:
目前我们确实确实这样的，类似爱快的方便的接口配置调整的功能。我们未来应该回做成一个插件来支持。目前固件肯定是有这个能力的，毕竟他是一个 标准的 OpenWRT 固件，这个功能必须有而且很强大，当然使用门槛也高：

[image]

closed 2022-04-24 by jannson

horse02, 2022-04-24:
按照你的方法设置，没用啊，我添加的新接口，换用设置LAN2口的几个协议，全部上不了网，这个是不是BUG呢，我以前OPWRT里的LAN口是很好设置的，请看我OP里设置的图，很方便
[image]
。