istoreos/istoreos #59: 求增加一个家长控制插件

RichKyle, 2022-04-24:
求一个家长控制插件！！！

jannson, 2022-04-24:
目前有 KoolProxy 插件可以做到。但是还没精力来维护。努力增加吧，未来。。。

jannson, 2022-04-24:
这个插件貌似可以办到：https://github.com/destan19/OpenAppFilter

RichKyle, 2022-04-24:
> 这个插件貌似可以办到：https://github.com/destan19/OpenAppFilter
> 
> 但是这个插件要求：
> 
> 1. 将应用过滤设备做主路由
> 2. 关闭软硬加速、广告过滤、QOS、多WAN等涉及到nf_conn mark的模块
> 3. 开启应用过滤并选择需要过滤的app即可生效
> 
> 估计无法集成，对周边插件产生影响。

我没安装成功…之前koolshare有一个家长控制挺好用的  不知道能否迁移过来

jannson, 2022-04-24:
你是真的想防止小孩子玩游戏，看抖音，还是想去掉一些广告啥的。

RichKyle, 2022-04-24:
> 你是真的想防止小孩子玩游戏，看抖音，还是想去掉一些广告啥的。

主要是为了控制孩子浏览视频和游戏的时间…家长控制之前可以设置时间段

jannson, 2022-04-25:
好的，那插件已经立项了。。。问题不大。

RichKyle, 2022-04-25:
> 好的，那插件已经立项了。。。问题不大。

感谢！！！！！

cnfjice, 2022-06-18:
[上网时间控制luci-app-accesscontrol_1-11_all.zip](https://github.com/linkease/istoreos/files/8932962/luci-app-accesscontrol_1-11_all.zip)