istoreos/istoreos #65: miniupnp有问题的更换版本可以解决。

lingyunzhiss, 2022-04-24:
系统自带的MiniUpnp不工作，更换低版本的解决了。
[image]
[image]

jannson, 2022-04-24:
感谢提供重要的信息，貌似有个用户也反馈过，当时没注意。

longaga, 2022-04-25:
[image]
没有问题吧，勾选启动 UPnP 与 NAT-PMP 服务

juzixs, 2022-04-25:
我的MiniUpnp也不工作 试试这个降MiniUpnp版本看看有没有效果去

jannson, 2022-05-06:
找了下，可能跟下面的两个 issue 相关：
https://github.com/openwrt/packages/issues/14145

https://github.com/openwrt/packages/pull/16011

貌似因为你们没公网 IP 导致？

syj119, 2022-05-06:
没有公网IP是大势所趋，越来越多的人upnp会不能用

jannson, 2022-05-07:
通过强制转发，可以让 UPNP 生效。目前强制转发已经成为默认配置项，如果是老配置文件，可以考虑自己配置，新刷的固件将会使用默认值。

closed 2022-05-07 by jannson

m0eak, 2023-03-25:
> 通过强制转发，可以让 UPNP 生效。目前强制转发已经成为默认配置项，如果是老配置文件，可以考虑自己配置，新刷的固件将会使用默认值。

开启强制转发还是不行。。。

ray0019, 2023-04-06:
> 问题已经确认，有朋友用一下老版本才能解决：
> 
> [image]

请问去哪里下载老版本的，最新0331版的固件自带的upnp还是无法启动。