istoreos/istoreos #76: esxi只要编辑istore虚拟机，就会提示请输入大于原始容量的磁盘大小。

yaoshuangshl, 2022-04-27:
如题，从官网下载img格式的镜像文件，转换成两个vmdk文件导入现有磁盘，开机正常启动，但是只要编辑虚拟机，
就会提示请输入大于原始容量的磁盘大小。并且保存按钮此刻为灰色的，但如果手动调整磁盘大小，依旧会报错并无法保存

jannson, 2022-04-28:
我们有直接提供的 vmdk 文件，能否使用？

yaoshuangshl, 2022-04-28:
直接停供的vmdk文件也是如此，并且无法打开电源，不过这个请输入大于原始容量的磁盘大小的问题我已解决，使用ssh到esxi上，vmkfstools -X 500M（需要扩容的大小） openwrt.vmdk（文件名），扩容后就不会报错了，但别的op镜像不需要这一步操作···

jannson, 2022-04-28:
[image]

DreamTimeONE1, 2022-05-01:
这个需要改下硬盘的容量 vmkfstools -X 500M（需要扩容的大小） openwrt.vmdk（文件名）,我用别的作者写的OP也有这个问题 ,不知道是不是OP编译的问题

zzk1996, 2022-05-10:
op,爱快在esxi里都有这情况，放心改就是，比如原先的硬盘大小为1023.6889,.你往大了取整，改成1024就可以了，要不然不让保存

jannson, 2022-07-08:
此问题终于修复了。因为我们的固件根目录，跟转换工具不兼容。我们改了点东西，现在好了！

WYC-2020, 2023-02-13:
能说下是哪里么