istoreos/istoreos #79: 卸载iStoreOS

wjx8881, 2022-04-27:
/usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.controller.jellyfin' from file '/usr/lib/lua/luci/controller/jellyfin.lua':
	/usr/lib/lua/luci/controller/jellyfin.lua:1: unexpected symbol
stack traceback:
	[C]: in function 'loader'
	/usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1107: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:1207: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:695: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:796: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

jannson, 2022-04-28:
rm -rf /tmp/luci-*

试试

wjx8881, 2022-04-28:
已经给我朋友给修复死了 我装了pve 想r86s 装爱快拨号 istpreos 旁 虚拟机里无法修改istoreso的网关似乎没有哪一项

发自我的iPhone

wjx8881, 2022-04-28:
PVE虚拟机 爱快主istoreos 旁路由 爱快开启跨三层应用显示异常 ist Oreos需要开启或者设置什么地方吗

发自我的iPhone

wjx8881, 2022-04-30:
直接在线升级到v20220429版本后 一直提示DNS错误且无法再上网 试过多种或者自动获取还是一样报错

jannson, 2022-05-18:
修改为阿里云 DNS，我相信肯定可以的。如果还不可以，那肯定是你的流量被劫持了。

closed 2022-05-18 by jannson