istoreos/istoreos #101: 请问softether vpn怎么调呢？

Blacklw, 2022-05-02:
我在istoreos的软件包里面安装了几个softether vpn5的软件，然后在server manager里面调好了设置，并且在流量规则里面建立了对应端口，但是在测试拨号的时候会提示服务器没有回应，这个问题应该怎么调呢？

jannson, 2022-05-05:
这个我们来测试下哈。我放到待确认问题里面。

fencatn, 2022-05-05:
防火墙需开放992 1194 5555端口，同时需要在命令行确认下 netstat -netlp | grep vpnserver 是否能看到这3个监听端口

jannson, 2022-05-05:
经过确认，这个不是问题。可能你设置的防火墙问题。

closed 2022-05-05 by jannson

Blacklw, 2022-05-06:
> 经过确认，这个不是问题。可能你设置的防火墙问题。

你好，我防火墙里面试过改几次不同的配置都不行，能给我看看您的设置吗？麻烦您了。另外我连接的是L2TP使用共享密钥去连接的。

Blacklw, 2022-05-06:
> 经过确认，这个不是问题。可能你设置的防火墙问题。

你好，我防火墙里面试过改3次不同的配置都不行，能给我看看您的设置吗？麻烦您了。另外我连接的是L2TP使用共享密钥去连接的。以下是我试过的配置组合
[image]
[image]
[image]

再补充几张我配置过与关于softether vpn的配置
[image]
[image]

还有就是和openwrt和merlin配置softether vpn的时候比较，istore在创建本地网桥的时候会多一个说明出来，我不知道是不是和这个有关系
[image]

Blacklw, 2022-05-06:
> 防火墙需开放992 1194 5555端口，同时需要在命令行确认下 netstat -netlp | grep vpnserver 是否能看到这3个监听端口

您说的我试过了，也确定端口是有监听状态的，但是还是一样的状态。
[image]

fencatn, 2022-05-17:
你用电脑拨下VPN有没有什么报错，如果到不了验证用户名密码那一步，说明防火墙设置有问题，如果提示验证失败那很可能是SOFTETHER设置的问题。
网上找了个攻略你参考下呢https://www.iloft.xyz/archives/softether-l2tp-openwrt.html 你也可以搜下视频，网上很多的。
最容易出错的地方，一个就是VPN网桥，一个就是防火墙

Blacklw, 2022-05-17:
> 你用电脑拨下VPN有没有什么报错，如果到不了验证用户名密码那一步，说明防火墙设置有问题，如果提示验证失败那很可能是SOFTETHER设置的问题。 网上找了个攻略你参考下呢https://www.iloft.xyz/archives/softether-l2tp-openwrt.html 你也可以搜下视频，网上很多的。 最容易出错的地方，一个就是VPN网桥，一个就是防火墙

[image]
这个图片就是报错额，我也不知道什么原因，最后我找到的解决办法是不装softethervpn5的所有服务和依赖，就单纯安装softethervpn和相关依赖，其它设置完全一样的，就好了，能连接上

soulofhan, 2024-03-05:
Blacklw，你好，你的istore能使用softether作用服务端组建虚拟局域网吗？请问下需要安装那些软件包以及进行哪些设置？谢谢