istoreos/istoreos #114: 考虑支持用户的软件中心的软件源的检查，保证能访问

jannson, 2022-05-06:
离线包安装失败，是因为用户的网络无法访问软件源，从而安装失败，倒是用户不知所措。考虑增加源的检查，不能访问则增加感叹号，让用户用向导去更新软件源。可以考虑跟网络检查放一起，如下图所示：

[image]

houzi-, 2022-05-07:
我觉得哦可以在插件安装日志里加入安装报错提示，如果无法安装怎直接输出日志引导用户去修改软件源，这样直接明了

genesis-lua, 2022-05-07:
> 离线包安装失败，是因为用户的网络无法访问软件源，从而安装失败，倒是用户不知所措。考虑增加源的检查，不能访问则增加感叹号，让用户用向导去更新软件源。可以考虑跟网络检查放一起，如下图所示：
> 
> [image]

如果可以的话，还是希望能把GitHub上的一些软件合集和kmod组成istore的软件库，在软件包里按需自取就好

jannson, 2022-05-09:
你说得这些，后续空闲了肯定会来做的。

jannson, 2022-05-18:
软件源错误，已经支持，并且在首页还支持软件源修改。

[image]

closed 2022-05-18 by jannson

gintokighost, 2022-06-09:
这个软件源怎么切换或是配置