istoreos/istoreos #145: I217-V网卡偶现断连

Pg0dxY, 2022-05-15:
偶尔游戏+下载视频/软件，负载比较大的时候会出现，表现为突然断网10秒左右随后恢复
技嘉B85N-Phoenix的主板，应该是I217-V的千兆网卡做WAN（eth0），0429版本的系统
查了下其他论坛似乎是驱动crash之类的问题（具体不太清楚，不知道有没有遇到类似问题的大佬补充下）
从那个帖子学到的解决方式是“ethtool -K eth0 tx off rx off”，用这个命令关掉一个offload选项，目前关闭后稳定运行了5天

日志的话从出问题到恢复连接如下：
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang:
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   TDH                  <35>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   TDT                  <62>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   next_to_use          <62>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   next_to_clean        <34>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] buffer_info[next_to_clean]:
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   time_stamp           <10083fa86>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   next_to_watch        <3b>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   jiffies              <10083ffd0>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553]   next_to_watch.status <0>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] MAC Status             <80083>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] PHY Status             <796d>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] PHY 1000BASE-T Status  <3800>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] PHY Extended Status    <3000>
Tue May 10 23:43:42 2022 kern.err kernel: [34903.819553] PCI Status             <10>
Tue May 10 23:43:43 2022 kern.err kernel: [34905.039360] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
Tue May 10 23:43:43 2022 daemon.notice netifd: Network device 'eth0' link is down
Tue May 10 23:43:43 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue May 10 23:43:43 2022 daemon.notice netifd: wan (30180): udhcpc: received SIGTERM
Tue May 10 23:43:43 2022 daemon.notice netifd: wan (30180): udhcpc: unicasting a release of 192.168.1.8 to 192.168.1.1
Tue May 10 23:43:43 2022 daemon.notice netifd: wan (30180): udhcpc: sending release
Tue May 10 23:43:43 2022 daemon.notice netifd: wan (30180): udhcpc: entering released state
Tue May 10 23:43:43 2022 daemon.notice netifd: wan (30180): Command failed: Permission denied
Tue May 10 23:43:43 2022 daemon.notice netifd: Interface 'wan' is now down
Tue May 10 23:43:43 2022 daemon.info avahi-daemon[6259]: Withdrawing address record for 192.168.1.8 on eth0.
Tue May 10 23:43:43 2022 daemon.info avahi-daemon[6259]: Leaving mDNS multicast group on interface eth0.IPv4 with address 192.168.1.8.
Tue May 10 23:43:43 2022 daemon.info avahi-daemon[6259]: Interface eth0.IPv4 no longer relevant for mDNS.
Tue May 10 23:43:43 2022 kern.info kernel: [34905.222441] e1000e: eth0 NIC Link is Down
Tue May 10 23:43:43 2022 daemon.notice netifd: Interface 'wan' is disabled
Tue May 10 23:43:43 2022 daemon.notice netifd: Interface 'wan' is enabled

houzi-, 2022-05-17:
目前I217-V这个版本驱动在OP下的通病，暂时只能通过hotplug来去启动执行offload选项，会在接下来的版本中加入的

jannson, 2022-05-18:
尝试打开这个？
[image]