istoreos/istoreos #151: 0518版本升级后，首页的互联网连接时间和IPV4及IPV6地址无法显示

milhu, 2022-05-18:
群里小伙伴也有相同问题，前面几版升级都未发现问题
[image]
。

gkl1368, 2022-05-18:
做旁路的时候，这里如何显示ipv6地址？谢谢

bailliesun, 2022-05-18:
我也遇到了同样的问题

nemo830, 2022-05-18:
我的也是 ipv6不显示 还有16g内存爆满

milhu, 2022-05-19:
宝哥他们已经查过我机子上的后台了，找到问题点了，应该会在下个版本修复的。

fulracoco, 2022-05-19:
同遇到了。 重启了好次，才正常。 拨号成功了， 有设备不能上网，而有的设备能上网， dhcp分配页面也没有看到设备，
op自身却也不能联网。

milhu, 2022-05-19:
> 同遇到了。 重启了好次，才正常。 拨号成功了， 有设备不能上网，而有的设备能上网， dhcp分配页面也没有看到设备， op自身却也不能联网。

我的只是单纯的不显示，但是不影响使用，跟你描述的情况还不尽相同

jannson, 2022-05-19:
应该今天修复。

gkl1368, 2022-05-19:
> 应该今天修复。

作为旁路 单lan口，还是不显示。

molin830, 2022-05-19:
> 群里小伙伴也有相同问题，前面几版升级都未发现问题 [image] 。

> 应该今天修复。

大神 518 519 开机一会16g内存就爆满
[image]

jannson, 2022-05-23:
内存爆满是因为内存跑到 buff/cache 里面了，目前原因未知。

jannson, 2022-05-25:
已经修复。

closed 2022-05-25 by jannson

gkl1368, 2022-05-26:
> 已经修复。

最新的0525还是不显示v6地址