istoreos/istoreos #167: 升级0519版本后，RTL8125网卡只能协商1Gbps

Pg0dxY, 2022-05-22:
升级0519版本后，RTL8125B网卡只能协商1Gbps
尝试使用“ethtool -s eth2 speed 2500”命令后，提示"Cannot advertise speed 2500"
ethtool结果如下↓
[image]

尝试重新安装0519版本不解决问题，降级安装到老版本一切正常

Pg0dxY, 2022-05-22:
补充配置
E3-1265L v3
技嘉B85N Phoenix
WAN口板载I-217V
LAN为一块四口2.5G网卡，四个独立的RTL8125B网卡

此前正常时使用的是0429版本

jannson, 2022-05-23:
收到。

jannson, 2022-05-28:
请尝试昨天发布的 0526 版本，增加了很多关于 USB 的驱动。我估计可以了。

Pg0dxY, 2022-05-28:
> 请尝试昨天发布的 0526 版本，增加了很多关于 USB 的驱动。我估计可以了。

在另外一台设备上尝试了0526版本，还是只能协商到1Gbps（图中的eth0）
[image]

尝试ethtool -s 命令强制协商2.5Gbps，提示cannot advertise
[image]

我这个卡是RTL8125B（和一些主板板载的是同款），和USB的8156/8152驱动似乎还不太一样，大佬有空的话可以再Check下

Acocoplay, 2022-06-13:
同RTL8125B网卡，我是安装了0610最新版，同样也只能协商在1Gbps

jjm2473, 2023-02-22:
没升级r8152驱动导致的

closed 2023-02-22 by jjm2473