istoreos/istoreos #169: 升级0519版本后主页显示内存占用99%

wildtigerw, 2022-05-22:


molin830, 2022-05-22:
内存占用升高就降不下来 还有 在线升级后 smb网络共享再也打不开

molin830, 2022-05-22:
[image]

molin830, 2022-05-22:
[image]

molin830, 2022-05-22:
518 519 都有这个问题

jannson, 2022-05-23:
top 看看，哪个进程内存占用特别高？

molin830, 2022-05-23:
> top 看看，哪个进程内存占用特别高？
大大 你帮忙看一下 我看不懂这么深奥你东西
[image]

wildtigerw, 2022-05-23:
我刚刚看了一下，丢个BT种子给qbittorrent下载后，占用率就一直保持99%。下载完毕后占用也不会自动清理。

jannson, 2022-05-23:
内存确实都自动到 buff/cache 里面了。如果手动释放内存倒是可以恢复正常。目前不知道这个是内核故意为之，还是其它啥问题。等我们确认。

jannson, 2022-05-23:
> 我刚刚看了一下，丢个BT种子给qbittorrent下载后，占用率就一直保持99%。下载完毕后占用也不会自动清理。

刚才用你的这个方法重现了这个问题：

[image]

从这个状态来看，内存是没问题的，只是放到了缓存里。这个是新版本的官方行为，实际上可用内存还很多，只是缓存起来，让内存跑得更快了。这个我们会更改前端的内存占用，让它减去可用数量，得到真实的内存使用情况。

jannson, 2022-05-23:
对了，有发现新问题，大家看看出问题的时候，有没有类似我下面的情况？

```
[ 5088.682693] usb 2-1: USB disconnect, device number 2
[ 5088.684799] blk_update_request: I/O error, dev sdb, sector 13879296 op 0x1:(WRITE) flags 0x0 phys_seg 201 prio class 0
[ 5088.696513] EXT4-fs warning (device sdb1): ext4_end_bio:315: I/O error 10 writing to inode 28114953 (offset 991952896 size 1048576 starting block 1735168)
[ 5088.696522] Buffer I/O error on device sdb1, logical block 1731072
[ 5088.703257] Buffer I/O error on device sdb1, logical block 1731073
[ 5088.709922] Buffer I/O error on device sdb1, logical block 1731074
[ 5088.716618] Buffer I/O error on device sdb1, logical block 1731075
[ 5088.723359] Buffer I/O error on device sdb1, logical block 1731076
[ 5088.730017] Buffer I/O error on device sdb1, logical block 1731077
[ 5088.736458] Buffer I/O error on device sdb1, logical block 1731078
[ 5088.743020] Buffer I/O error on device sdb1, logical block 1731079
[ 5088.749655] Buffer I/O error on device sdb1, logical block 1731080
[ 5088.756118] Buffer I/O error on device sdb1, logical block 1731081
[ 5088.762712] JBD2: Detected IO errors while flushing file data on sdb1-8
[ 5088.762715] blk_update_request: I/O error, dev sdb, sector 5859751608 op 0x1:(WRITE) flags 0x800 phys_seg 5 prio class 0
[ 5088.773966] Aborting journal on device sdb1-8.
[ 5088.778641] blk_update_request: I/O error, dev sdb, sector 5859735552 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
[ 5088.789892] Buffer I/O error on dev sdb1, logical block 732463104, lost sync page write
[ 5088.798199] JBD2: Error -5 detected when updating journal superblock for sdb1-8.
[ 5094.446710] EXT4-fs error (device sdb1): __ext4_find_entry:1532: inode #28114945: comm aria2c: reading directory lblock 0
[ 5155.405345] EXT4-fs error (device sdb1): __ext4_find_entry:1532: inode #28114945: comm aria2c: reading directory lblock 0
[ 5215.846862] EXT4-fs error (device sdb1): __ext4_find_entry:1532: inode #28114945: comm aria2c: reading directory lblock 0
[ 5276.299838] EXT4-fs error (device sdb1): __ext4_find_entry:1532: inode #28114945: comm aria2c: reading directory lblock 0
[ 5308.233437] EXT4-fs error (device sdb1): ext4_journal_check_start:61: Detected aborted journal
[ 5308.242960] EXT4-fs (sdb1): Remounting filesystem read-only
[ 5308.249100] ------------[ cut here ]------------
```

jannson, 2022-05-25:
最新版本修复了此问题。

closed 2022-05-25 by jannson