istoreos/istoreos #182: 拨号上网会显示DNS错误

elbert1996, 2022-05-26:
拨号上网的时候隔一段时间就检测提示DNS错误 然后就不能上网了  但是过几秒种后又好了 然后过一会又重复提示错误 使用的版本号为
OpenWrt 21.02.2 2022052508

jannson, 2022-05-28:
尝试修改，如图所示打勾，看看：
[image]

wildtigerw, 2022-05-28:
遇到同样问题，按老大的方法修改了，再体验一阵试试。

sonrinhou, 2022-05-28:
同样的问题，不停地提示DNS配置错误[image]

elbert1996, 2022-05-30:
这个选项打钩后网页打不开了

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F4100344%2F170807817-daaea76e-21b7-480a-92c0-e871e0a12c08.png&data=05%7C01%7C%7C701293acee5249a9c2fe08da4058f9d8%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637893048148770614%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=OPYLrA8hKwjo6TxR9omPZJsVKqQFUyasrRKAyXB%2FZbo%3D&reserved=0>