istoreos/istoreos #190: 能不能添加tar.gz格式包，方便在docker中使用

laye0619, 2022-05-28:
RT，感谢！

jannson, 2022-05-30:
英雄所见略同啊，刚刚做了一个 Docker 的项目：https://github.com/linkease/istoreos-in-docker

我马上提交 rootfs 上来。

laye0619, 2022-05-30:
太感谢了！

jannson, 2022-05-31:
https://fw.koolcenter.com/iStoreOS/alpha/openwrt-x86-64-generic-squashfs-rootfs.img.gz

昨天放这里了，忘记说了。

laye0619, 2022-05-31:
大佬，多问一句，我知道rootfs.tar.fz可以直接在dockerfile里面ADD就好了，这种rootfs.img.gz我怎么做成docker镜像？

jannson, 2022-06-15:
搞半天研发给错的文件给我了，我以为 OK 了我都没注意了，不好意思。我等会再上传一个。。。

laye0619, 2022-06-16:
> 搞半天研发给错的文件给我了，我以为 OK 了我都没注意了，不好意思。我等会再上传一个。。。

上传了告诉我一下哈，多谢！

closed 2022-06-27 by jannson

gzygood, 2023-03-22:
@jannson 大哥，请最新版的.tar.gz  我想用pve的lxc容器安装，直接虚拟机占资源啊

jjm2473, 2023-03-22:
注意使用.tar.gz的话，没办法升级。

gzygood, 2023-03-22:
好的， 大不了到时候删了容器重建，istoreos是旁路
那这么说我没搞错，/dev/loop1p2      263168  787455  524288  256M 83 Linux
这个分区是正确的？我一直以为是那个2G大小的，就是一直死活都mount不上

jjm2473, 2023-03-22:
@gzygood  2GB的是overlay分区。https://openwrt.org/docs/guide-user/additional-software/extroot_configuration

系统文件是squashfs打包的，只读，如果要写入文件，需要使用overlayfs将差异文件写到overlay分区。所有用户修改的东西都会放在overlay分区，这也就是为什么openwrt能够支持恢复出厂设置，只要删掉overlay里面的文件就行了。

jjm2473, 2023-03-22:
@gzygood  2GB的分区在第一次开机前是空的，没有任何数据。

gzygood, 2023-03-22:
好的，感谢，我测试下配置一些跟之前用虚拟机安装的软件包等等，看看有没问题
完全小白，一直在摸索。再次感谢解答~

jjm2473, 2023-03-22:
升级和恢复出厂设置可能都会失效，或者产生不可预见的行为（例如把宿主的硬盘设备文件映射到容器里，升级时把宿主机的硬盘给刷了）。

gzygood, 2023-03-22:
现在已经发现了，istoreos里面内存、硬盘啥的都是直接显示的宿主机的全部占用和空闲

gzygood, 2023-03-22:
iStoreOS有自动更新功能吗？如果有的话 ，请问要如何关闭呢。我看首页里有个固件更新按钮一直在转圈圈~

XYZliang, 2023-08-05:
大哥们有最新版的.tar.gz吗，想要用lxc容器