istoreos/istoreos #193: 局域网内互访SMB共享文件夹很慢

tommytianhm, 2022-05-29:
固件版本：OpenWrt 21.02.2 2022051913
运行环境：N5105软路由，Esxi 7.0.2 虚拟机，OpenWrt为主路由
window下访问其他机器SMB共享的文件夹还算正常，但是iOS或安卓下的app（比如Nplayer）打开共享文件夹就很慢。
主路由切换为iKuai访问同样的共享文件夹就很顺畅。
不明原因

jannson, 2022-05-31:
有个 Samba 兼容模式？试试。不行到时候再到下个版本测试。我们最近在优化了一些性能的东西。

jannson, 2022-06-09:
最近优化了一些性能，请继续用最新版本测试下，谢谢。

FFTU, 2022-06-10:
> 固件版本：OpenWrt 21.02.2 2022051913 运行环境：N5105软路由，Esxi 7.0.2 虚拟机，OpenWrt为主路由 window下访问其他机器SMB共享的文件夹还算正常，但是iOS或安卓下的app（比如Nplayer）打开共享文件夹就很慢。 主路由切换为iKuai访问同样的共享文件夹就很顺畅。 不明原因

你好，ios下Nplayer 的SMB 理论上应该是多少的速度？我win之间可以100M/s，但是Nplayer连路由（AX86U）5G，只有40左右的速度（iPhone 12）

tommytianhm, 2022-06-11:
> 最近优化了一些性能，请继续用最新版本测试下，谢谢。

升级到了OpenWrt 21.02.2 2022061019，访问其他OpenWrt机器的共享smb依然很慢。
通过cifs挂载这个smb共享，然后通过这个istoreOS（主路由）通过smb服务共享出去，访问就正常了，现阶段只能通过这样解决了，有点麻烦。

tommytianhm, 2022-06-11:
> 
你说的速度Byte，还是bit？差了8倍哦
应该是受限于wifi的速度吧

tommytianhm, 2022-06-11:
> > 固件版本：OpenWrt 21.02.2 2022051913 运行环境：N5105软路由，Esxi 7.0.2 虚拟机，OpenWrt为主路由 window下访问其他机器SMB共享的文件夹还算正常，但是iOS或安卓下的app（比如Nplayer）打开共享文件夹就很慢。 主路由切换为iKuai访问同样的共享文件夹就很顺畅。 不明原因
> 
> 你好，ios下Nplayer 的SMB 理论上应该是多少的速度？我win之间可以100M/s，但是Nplayer连路由（AX86U）5G，只有40左右的速度（iPhone 12）

你说的速度Byte，还是bit？差了8倍哦
应该是受限于wifi的速度吧

tommytianhm, 2022-07-02:
今天发现是我的smb服务器设置网关有问题，不是iStoreOS的问题，错怪了，抱歉。
这个问题关闭吧

closed 2022-07-02 by tommytianhm