istoreos/istoreos #226: vpn请增加些常用款

chfjxmtable, 2022-06-15:
希望vpn除提供openvpn sever外，也能有ipsect和pptp

jannson, 2022-06-15:
OK，ipset pptp 考虑上软件中心。。。

chenxi520, 2022-06-19:
希望加入softether vpn   软件包安装后无法使用

> OK，ipset pptp 考虑上软件中心。。。

希望加入softether vpn   软件包安装后无法使用

chenxi520, 2022-06-19:
> OK，ipset pptp 考虑上软件中心。。。

希望加入softether vpn   软件包安装后无法使用

zhangtaidong, 2022-07-17:
softethervpn 希望上软件中心

TshineZheng, 2022-07-23:
同求 softethervpn

rockyzhz, 2022-09-07:
softethervpn 。。。今天才发现居然softethervpn也没有。。。我还以为是标配了呢，啥时候能出来呀？在考虑先换个固件了

chfjxmtable, 2022-09-08:
用wireguard和openvpn

xyaolong, 2022-10-27:
强烈要求增加softethervpn，用习惯了，并且稳定！

byteinfo, 2022-11-10:
softethervpn-server 在软件包安装了也无法使用，希望上软件中心

byteinfo, 2022-11-10:
> softethervpn-server 在软件包安装了也无法使用，希望上软件中心

Phoenixiao191, 2022-12-28:
同求，openvpn连上了，访问不了内网设备，我也是心累

chenxi520, 2023-04-25:
> 同求，openvpn连上了，访问不到内网设备，我也是心累

已经看openwrt官方资料在这个固件成功安装使用

zhshum, 2023-06-29:
opkg install wireguard 提示没有安装包，可以增加下wireguard吗

miRemid, 2023-12-22:
netbird希望也能支持一下

chfjxmtable, 2024-01-24:
> 没有PPTP类的VPN软件可以用吗？大家现在都是怎么用的？

现在最方便设置与使用的是tailscale

soulofhan, 2024-03-06:
各位大佬们
我用docker直接搭建softether端能成功使用，不过这样子需要做的NAT转换太多了
链接上之后，我用客户机直接打开192.168.100.1是可以进去的
以下是整条链路
istore 192.168.100.1 （docker桥接：172.17.0.0/16）-->docker 容器ip 172.17.0.2（softether NAT DHCP地址池192.168.30.0/32）-->客户机开启openvpn链接成功后获取的 ip地址：192.168.30.13/32（期望该ip地址能直接从istroe获取）

大神们有没有办法让客户机的ip直接从istore的DHCP服务中获取192.168.100.x
或者让docker容器的ip也是192.168.100.0网段下？

closed 2025-05-21 by jjm2473