istoreos/istoreos #239: 后台管理界面字体映射错误

wydinhk, 2022-06-18:
Argon 主题默认西文字体是 Google Sans，在其他路由器固件上显示正确，但在 iStoreOS 中映射错误，似乎字体文件没有加载。在苹果设备上所有西文一律显示为苹方，看起来比较奇怪。Logo 所使用的 TypoGraphica 字体也是如此情况。希望可以排查一下。

作为临时解决方案，我自己是在 /www/luci-static/argon/css/cascades.css 中修改了一些内容，在 --font-family-sans-serif 项最前面加上 `-apple-system,system-ui,BlinkMacSystemFont,`，调用苹果系统原生字体 San Francisco 显示，效果不错。另外出现 TypoGraphica 的地方，在其后面也作如此添加。

注意到首页网速监测图表右上角速度显示样式在 /www/luci-static/quickstart/style.css 中，需要单独修改。而图表内部光标悬置时出现的时间、速度信息不知道如何修改。

以下为更改前后效果对比：

[image]

[image]

[image]

<i[image]

[image]

[image]

jannson, 2022-06-21:
专业啊兄弟，我看不懂，我让我们工程师看看。

jjm2473, 2023-02-18:
如果是argon主题的问题，可以去argon项目反馈

closed 2023-02-18 by jjm2473

akira304moto, 2023-02-18:
正文第一段已经说了“Argon 主题的Google Sans 在其他路由器固件上显示正确”，不是 Argon 主题的问题。所以才比较奇怪。

jjm2473, 2023-02-19:
哦，我想起来了，是openwrt的css压缩工具的bug，会导致多个 `@font-face` 被删除到只剩一个，在Chrome上显示又没问题所以没关注。

相关bug的链接 https://github.com/openwrt/luci/issues/5067 https://github.com/openwrt/luci/issues/5755

我看看加上 `LUCI_MINIFY_CSS:=0` 能不能解决。

jjm2473, 2023-02-19:
加上`LUCI_MINIFY_CSS:=0`就修复了。没发现网速监测图的样式有问题，估计之前已经修复了