istoreos/istoreos #240: R2S: frp server/ frp client RPCError

XuyangSHEN, 2022-06-18:
主机名 | iStoreOS
型号 | FriendlyElec NanoPi R2S
架构 | ARMv8 Processor rev 4
Target Platform | rockchip/armv8
固件版本 | OpenWrt 21.02.2 2022061017 / LuCI openwrt-21.02-cn branch git-22.160.13453-cb18896
内核版本 | 5.4.179

系统安装版本： [openwrt-2022061017-nanopi-r2s-squashfs.img.gz](https://fw.koolcenter.com/iStoreOS/r2s/openwrt-2022061017-nanopi-r2s-squashfs.img.gz)
iStore版本 [v0.1.10-4]

从iStore 安装frp服务器和frp客户端，都出现了 RPCError

RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply

各位大佬有什么办法吗？

jannson, 2022-06-27:
不好意思，之前没注意，等我们确认下。

kuangxz, 2022-07-17:
[image]
现在我也出现了这个问题，大佬解决了么