istoreos/istoreos #254: 无法识别2.5g网口

GTXGOGOGO, 2022-06-23:
我从eSir的高大全过来的刷了istoreOS——6.10日的版本，j4125软路由esxi6.7，螃蟹网卡，进入固件后接口只显示1000m，之前的高大全op正常显示2.5G，请问是什么原因，还有我的三个网口eth0，1，2，0号是iptv，1号lan口（接着2.5g交换机），2号是wan口，首页网络只显示iptv口网络情况，如何调整为显示wan口也就是eth2网络情况

GTXGOGOGO, 2022-06-23:
网卡是REALTEK8125B 2.5G网卡

jannson, 2022-06-24:
我映像中，这个网口我们是支持的。并且我这里有这个网卡的设备，也是支持的。

首页可能把 iptv 认为是 WAN，目前我们确实首页只认识一个 WAN，但不影响使用。

curtain12138, 2022-06-24:
我也是螃蟹的2.5G，也是用不了，协商速率只有1000M，继续用回esir的固件了，也有可能8125BG的可以用，8125B的协商不了，大概率还是驱动有问题吧。

Acocoplay, 2022-06-26:
我也是8125B，也是只能协商在1Gbps，0610之前好几个版本都是不能协商到2.5，好像只有刚支持这个网卡的后面几个版本才行

yahochina, 2022-07-05:
我一个8125B和8125不带B，都可以协商2.5G，但奇怪的是DHCP都可以获取地址和网关，但是PING 不通网关，上不了网。

wangweiweiwei, 2022-07-09:
Intel i225 2.5g主页也显示1000m测速是能上2.5gb的，但是是半双工

GavinIves, 2022-07-09:
我的也是8125b 能识别出2500的模式 但是最终协商的是1000 但我确认所有的链路都支持2500

jjm2473, 2023-04-11:
iptv的问题看这里 https://github.com/istoreos/istoreos/discussions/693

closed 2023-04-11 by jjm2473