istoreos/istoreos #256: iStoreOS 在即将的两波功能更新之后，计划以 fork OpenWRT 的方式开源啦

jannson, 2022-06-24:
大家对开源怎么看？欢迎留言！我先说我小宝个人的观点，不代表我们公司跟我们内部研发团队！

开源首选的好处，是让大家对 iStoreOS 内部的有更多细节的了解！更具体来说下：

好处：
* 又多了一个国人的固件，给大家带来更好的网络存储系统，让大家选择不同价位的硬件，都能有个相对固定的系统体验
* 遵循了相关开源协议，至少 youtube 的 UP 主做视频的时候，不被别人 BBB 了
* 更多人了解 iStoreOS 内部的细节，出现问题更及时的被排查
* 可能有人贡献优质的代码
* 可能有商业公司邀请别人贡献代码（商家可以自己请人做机型适配）
* 给更多人学习的机会

坏处：
* 更多人参与项目，并可能发布相似的延伸版本，导致了版本的混乱（就像 iStoreOS 自己从 OpenWRT 延伸一样，但是我们完全遵循 OpenWRT 的代码架构进行，我们修改的部分都是做成了 ipk 插件可以随时卸载）
* 更多极客想添加高级特性功能，反而导致版本不稳定
* 商业巨头 fork 之后换个壳还顺便踩一脚 iStoreOS，最后继承了 iStoreOS 的所有好处，让 iStoreOS 没有动力更新

顺便提一下，iStoreOS 对开发者的代码贡献有严格要求，原因来自于我们的目标：

* 因为是要给更小白的用户使用，必须保证稳定是第一要素
* 我们不需要功能复杂到普通用户用不上，所以最好是精简，复杂功能可以走软件中心
* 提交的代码，尽可能不影响到其他已上线的应用
* 并且要尽可能的跟 OpenWRT 主线分支兼容，一切代码提交都要 OpenWRT 标准

我们内部开发者也常常因为一点点代码是否放对了位置而争论很久，甚至影响了进度。

总之欢迎大家参与讨论，我相信 iStoreOS 未来会影响到大家。

KeplerSSSSS, 2022-06-30:
支持开源
希望宝哥能做好商店的项目,
因为iStoreOS对于我这种新手来说意义重大.
iStoreOS的出现让我发现了更多Openwrt的可玩性
很多方便的功能也能很快上手使用了
极大的改善和方便了我的生活
但是目前商店的内容还是远远不够的
如果开源项目能够加快内容的填充  我当然是大力支持的

最后  希望宝哥能给商店加一个XTeve 配合商店里的jellyfin使用

mlpzaq9998, 2022-07-01:
支持开源,刚好有个rk3399板子想用你的iStore，开源了再也不用自己拼凑代码了

jannson, 2022-07-20:
除了 RK3568 还合并中，其它已经完成。

closed 2022-07-20 by jannson