istoreos/istoreos #257: Frpc报错误代码4：资源未找到

XloadNetwork, 2022-06-24:
设备：FriendlyElec NanoPi R5S 友善R5S 4G内存+8G emmc
系统环境：r5s-update-0610.img On Emmc存储
固件版本：OpenWrt 21.02.1 2022061021 / LuCI branch git-22.161.50247-cb92cb0
内核版本：4.19.219
报错插件：frpc v1.0.0
安装方式：iStore [v0.1.10-4]软件商店自动安装
错误信息：
RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
at ClassConstructor.handleCallReply (http://192.168.8.1/luci-static/resources/rpc.js?v=git-22.161.50247-cb92cb0:15:3)
[image]

jannson, 2022-06-27:
我们试用下哈。。。

zhuzhe1983, 2022-07-02:
我可以插队问一下你是咋写入 emmc的么，我用那个瑞芯威升级磁盘工具做的tf卡只能启动tf卡上的系统，在那个安卓界面任何选项都无法写入emmc

zhuzhe1983, 2022-07-02:
最后用tf版本的友善openwrt自带emmc写入 搞定了。。

jannson, 2022-07-08:
我们教程里面，有个工具，可以刷入 emmc

cyqtmxk, 2024-07-04:
同样的问题，但是在系统自带的软件包里安装luci-app-frpc就没有问题，在istore里就出同样的错误。
系统: iStoreOS 21.02.3, r0-17b7b13\
[image]