istoreos/istoreos #265: x86的软路由单线多拨用不了

liucongx, 2022-06-29:
[image]

jannson, 2022-06-30:
感谢反馈，貌似多人都不行，估计是我们固件的 BUG。等我们开发者看看怎么回事。。。

wangweiweiwei, 2022-07-09:
me too，之前的版本多拨成功过

iovejieba, 2022-12-12:
一样，还没修复么？

iovejieba, 2022-12-12:
目前最新的固件依旧如此

kayboy69, 2023-05-13:
一年了还是这样

iovejieba, 2023-05-13:
> 一年了还是这样

可能...不在他们的计划修复....不行就换固件吧

kayboy69, 2023-05-13:
> > 一年了还是这样
> 
> 可能...不在他们的计划修复....不行就换固件吧

嗯，感谢回复

iovejieba, 2023-05-13:
> > > 一年了还是这样
> > 
> > 
> > 可能...不在他们的计划修复....不行就换固件吧
> 
> 嗯，感谢回复

它这个其实主要是 三方插件...mwan3  和多拨插件的问题   但可以修 不是能修  只是估计不在他们更新计划

iovejieba, 2023-05-13:
非刚需 人家就懒得搞了

kayboy69, 2023-05-13:
> 非刚需 人家就懒得搞了

好的，准备试试其他固件，感谢

jinzhongjia, 2023-12-17:
现在单线多拨基本能用也会失败，不用再尝试了

iovejieba, 2023-12-17:
> ![123](https://private-user-images.githubusercontent.com/22174761/291064657-665f9452-d677-43c2-8fc0-4eb559fa246c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI4MTk3OTUsIm5iZiI6MTcwMjgxOTQ5NSwicGF0aCI6Ii8yMjE3NDc2MS8yOTEwNjQ2NTctNjY1Zjk0NTItZDY3Ny00M2MyLThmYzAtNGViNTU5ZmEyNDZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE3VDEzMjQ1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFhMDAxNWY3NTA0NjM1MTk3NjBlZGFhMjE4ZTEzOWQ4ZGYxNDc4ZjJmNTA3MjBiMzY0NzdlM2ZkMTQ3ZTg4ZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.72SS4tNUpjaAa9XmIdObif7joPNJCcwpoYJ7RU2DBSg) 换固件了，成功了。。。

重启 可能会网关失败 巴拉巴拉的问题 导致mwan3报错   考虑稳定单线多拨换固件吧  这功能不是开发组刚需 不会更的  要更早更了

iovejieba, 2023-12-18:
> > > ![123](https://private-user-images.githubusercontent.com/22174761/291064657-665f9452-d677-43c2-8fc0-4eb559fa246c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI4MTk3OTUsIm5iZiI6MTcwMjgxOTQ5NSwicGF0aCI6Ii8yMjE3NDc2MS8yOTEwNjQ2NTctNjY1Zjk0NTItZDY3Ny00M2MyLThmYzAtNGViNTU5ZmEyNDZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE3VDEzMjQ1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFhMDAxNWY3NTA0NjM1MTk3NjBlZGFhMjE4ZTEzOWQ4ZGYxNDc4ZjJmNTA3MjBiMzY0NzdlM2ZkMTQ3ZTg4ZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.72SS4tNUpjaAa9XmIdObif7joPNJCcwpoYJ7RU2DBSg) 换固件了，成功了。。。
> > 
> > 
> > 重启 可能会网关失败 巴拉巴拉的问题 导致mwan3报错 考虑稳定单线多拨换固件吧 这功能不是开发组刚需 不会更的 要更早更了
> 
> 我这里常年一线多拨，新买了个N100，想试试这个版本的固件，结果最刚需的功能无法使用，其实可以修一下，还有好多地方是可以一线多拨的。

属作者非刚需项目，要更的话 不至于快两年没动静....大雕的内核的话  尽量找 22年11月1号前后的内核版本  多拨的话基本做主设备个把月不重启没什么问题 后续版本或多或少mwan3兼容和  dhcp 都会有随机性抽风问题 原因未知

lianyu2023, 2023-12-21:
> 。

换的哪个固件？为何我用最新的iStoreOS 22.03.5 2023121510，也是不行呢？

lianyu2023, 2023-12-21:
> ![123](https://private-user-images.githubusercontent.com/22174761/291064657-665f9452-d677-43c2-8fc0-4eb559fa246c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3Lm [...]

换的哪个固件？为何我用最新的iStoreOS 22.03.5 2023121510，也是不行呢？

iovejieba, 2023-12-21:
> 换的哪个固件？为何我用最新的iStoreOS 22.03.5 2023121510，也是不行呢？

换其他分支开发的别用iStoreOS 基本大部分都可以

lianyu2023, 2023-12-21:
> 换其他分支开发的别用iStoreOS 基本大部分都可以

我尝试了别的，还是不行，方便分享一下你用的固件吗？