istoreos/istoreos #272: socat用luci配不上

tolshao, 2022-07-01:
[image]
如图，命令行的方式，实测可以，但是用luci界面配置，一直都是×，大佬help

jannson, 2022-07-02:
之前第上次上架的时候我测试是可以的。那我们再不测试确认下。

tolshao, 2022-07-13:
补充一个信息。
1、无论luci怎么设置，luci显示都是×，ipv6可用，ipv4无效，“ps -ef | grep socat”之后，发现只监听了ipv6，和luci里配置对不上，ipv6是可以用的；
2、luci里停止启用后，系统进程无法被清除，需要手动kill，来回设置会占用好多好多端口；
3、配置文件里有socat、socat-opkg，是否调用不对，手动删除过这2个，重新安装无效；
[image]
[image]
[image]

tolshao, 2022-08-20:
偶然刷到了socat更新的消息，然后“升级插件”，无法使用；

tolshao, 2022-08-20:
删除插件，然后手动删除/etc/config/socat、socat_luci配置文件，重新安装插件（无报错）；
socat配置完后，未生效，还是×，用“ps -ef | grep socat”命令，无新进程生成。
目前只能用“系统/启动项"里的命令行自动启动

jannson, 2022-09-22:
最新软件中心的插件 OK 了，需要卸载再安装就行了。之前是因为他们的软件包搞冲突了。。。

closed 2022-09-22 by jannson

tolshao, 2022-09-23:
描述一下我还是不行的过程：
1、卸载socat，重装（报错什么配置文件重复），然后不能用；
2、卸载socat，手动ssh登录，删除/etc/config/socat*配置文件，重装（未报错），还是不能用；
更新插件中心至v0.1.12-7后，发现socat还是2022-08-25的版本

sooway, 2022-10-22:
我发现，在0924这个版本里，是可以正常使用的，然后在lede 3.2里是不能成功使用的