istoreos/istoreos #273: 【严重问题】非EFI固件写入U盘进行引导出现的BUG

bcseputetto, 2022-07-01:
最新版的(可能前几个版本也是有这个问题)非EFI固件使用rufus或者balenaEtcher写入U盘
在软路由设备硬盘内没有安装openwrt，是空硬盘或者无分区，的情况下。
引导至U盘会出现无法挂载overlay分区的情况，也就导致了/etc/config下的配置文件大部分丢失。
主机名也跟着一起丢失变成了最近流传广泛的 root@(none) 情况。
当软路由设备硬盘内含有一个完整的openwrt时，该问题不会重现。
efi版固件无论软路由设备硬盘是否已有openwrt，都不会重现该问题。
[image]
[image]

以上是通过物理机直接引导测试复现，该问题在使用Esxi等各类虚拟机均有出现。在虚拟机上可以通过扩大虚拟硬盘(磁盘)大小来解决这个问题。

2666, 2022-07-01:
遇到了同样的问题，题主是怎么解决的呀，我是exsi虚拟机，最开始是老版本的istore，后续在线升级，然后重启发现系统进不去，卡bios，索性重新创建新的虚拟机，结果遇到你以上的问题，传统bios引导最新的istoreos出现root（noon），而且包的大小不对，只有300多M

bcseputetto, 2022-07-02:
> 遇到了同样的问题，题主是怎么解决的呀，我是exsi虚拟机，最开始是老版本的istore，后续在线升级，然后重启发现系统进不去，卡bios，索性重新创建新的虚拟机，结果遇到你以上的问题，传统bios引导最新的istoreos出现root（noon），而且包的大小不对，只有300多M

虚拟机类安装的可以通过扩大虚拟硬盘/磁盘大小解决这个问题，我看他们Exsi一般把img转换成vmdk，然后在虚拟机上挂载该磁盘，大小改大后应该就可以解决

jannson, 2022-07-02:
感谢反馈，我把这个问题放到高优先级里面！

jannson, 2022-07-02:
这个问题的核心是，Ｕ　盘里面的引导，内系统原本在盘里面的系统，因为ＵＵＩＤ　一样，导致冲突了。还想办法解决中。

jjm2473, 2022-07-04:
打算按这个方案来解决 https://github.com/linkease/istoreos/issues/281

jannson, 2022-07-08:
现在镜像改了，用写盘工具，用最新的版本应该可以用了。

closed 2022-07-08 by jannson

jjm2473, 2022-07-27:
虚拟机镜像转换的问题已经修复。uuid冲突的问题，转到 https://github.com/istoreos/istoreos/issues/281