istoreos/istoreos #274: 建议新增功能 lan口下设备ipv6 的ddns

gzxiexl, 2022-07-02:
Ipv6下每个设备都有独立的ipv6 地址，目前网上只有依靠脚本实现该功能，建议宝大能够实现插件化。LAN口设备的ipv6地址都是向主路由请求的，同时加上防火墙放行自定义端口或者端口段就完事了。

<sub>Sent from <a href="https://apps.apple.com/cn/app/id1314212521">PPHub</a></sub>

jannson, 2022-07-02:
请提供对应的脚本，以及使用方法。

gzxiexl, 2022-07-02:
https://www.right.com.cn/forum/thread-668697-1-1.html 恩山论坛看到的，对小白来说太有门槛了，希望istoreOS能插件化造福小白

shaojs321, 2023-07-12:
+1，同样需要，我看好像只有爱快有这个功能，直接给下级设备的ipv6做ddns

github上ddnsgo项目给了一些想法，
https://github.com/jeessy2/ddns-go/wiki/%E9%80%9A%E8%BF%87%E5%91%BD%E4%BB%A4%E8%8E%B7%E5%8F%96IP%E5%8F%82%E8%80%83
通过   ip -6 route | awk '{print $1}' | awk '/240:?/' | awk -F::/ '{print $1 ":suffix of other mac"}'  命令获取ipv6前缀，再结合设备mac地址直接得出所有设备ipv6地址，这样就能申请ipv6域名了

haobinnan, 2024-01-12:
我做了类似的如果你想用 可以用我的接口

#!/bin/sh

MYIPV4=`ip addr show dev pppoe-wan | grep "inet " | awk '{ print $2 }'`
MYIPV6=`ip addr show dev br-lan | sed -e's/^.*inet6 \([^ ]*\)\/.*$/\1/;t;d' | head -n 1`

/usr/bin/curl -L "https://notepad.haobinnan.cn/ddns/aliyun_ddns.php\
?AccessKeyID=****************\
&AccessKeySecret=****************\
&DomainName=自定义.cn\
&IPv6PD=${MYIPV6}\
&SubDomainRecordCount=6\
&DR_1=A|@|${MYIPV4}\
&DR_2=AAAA|@|${MYIPV6}\
&DR_3=A|v4|${MYIPV4}\
&DR_4=AAAA|pc|FF:FF:FF:FF:FF:FF|MacToEUI64\
&DR_5=AAAA|server|FF:FF:FF:FF:FF:FF|MacToEUI64\
&DR_6=AAAA|kvm-pc|FF:FF:FF:FF:FF:FF|MacToEUI64"

参考吧。