istoreos/istoreos #275: iStore里能加个ipsec服务端吗

w9315273, 2022-07-02:
iStore里能加个ipsec服务端吗

jannson, 2022-07-02:
可以，需要等等。ｉｓｓｕｅ在这里，我们会更新支持。

johnhorse, 2023-04-02:
相同需求  现在还没有吗？

pengxuanzhi, 2023-06-22:
我也想用这个功能

luyufan498, 2023-10-25:
求支持

pengxuanzhi, 2023-10-25:
干嘛转发给我呀，真奇怪

CatttttttttttttttttttttttttttM ***@***.***> 于 2023年10月25日周三

soulofhan, 2024-03-06:
我用docker直接搭建softether端能成功使用，不过这样子需要做的NAT转换太多了
链接上之后，我用客户机直接打开192.168.100.1是可以进去的
以下是整条链路
istore 192.168.100.1 （docker桥接：172.17.0.0/16）-->docker 容器ip 172.17.0.2（softether NAT DHCP地址池192.168.30.0/32）-->客户机开启openvpn链接成功后获取的 ip地址：192.168.30.13/32（期望该ip地址能直接从istroe获取）

大神们有没有办法让客户机的ip直接从istore的DHCP服务中获取192.168.100.x
或者让docker容器的ip也是192.168.100.0网段下？

jazsm, 2024-10-10:
2022年就有人提出，现在都2024年了，希望能加上就好，因为这个ipsec不需要下载APP，iPhone手机自带客户端，非常好用。

w9315273, 2024-10-29:
> > > 2022年就有人提出，现在都2024年了，希望能加上就好，因为这个ipsec不需要下载APP，iPhone手机自带客户端，非常好用。
> > 
> > 
> > https://github.com/hwdsl2/docker-ipsec-vpn-server 用docker整吧
> 
> 因为是小白按步骤做了确链接不上，请问有视频教程吗

那里已经有非常详细的中文文档了……
我就给你简单说下 能不能行就看你自己了 还不明白就好好翻阅文档 当成是一个挑战来完成它
先创建个docker网络 专门给VPN用
`docker network create --driver bridge --subnet 172.18.0.0/30 --gateway 172.18.0.1 IKEv2`

然后看看这个新建的网口ID叫啥
`docker network ls`
你会看到类似内容 `48f0a288bc0c    IKEv2`
之后在op的`防火墙区域`添加一个`区域` 就叫 `IKEv2`
`全接受` 允许转发到`wan` `lan`
允许来自`lan`的转发
`高级设置` `涵盖的设备` 找找`br-48f0a288bc0c` 注意你的网口ID是不同的 参考前几句

端口转发
把`UDP 500` `UDP 4500` 都转发到 `IKEv2`区域的`172.18.1.2`

创建`NAT规则` `任何` `lan` `任意` `任意` `MASQUERADE`
`高级设置` `额外参数` `-i br-48f0a288bc0c` 注意你的网口ID是不同的 参考前几句

创建个环境变量文件 `vi .env` 写入以下内容
```
VPN_IKEV2_ONLY=yes # 只使用IKEv2模式
VPN_DNS_NAME=你域名
VPN_DNS_SRV1=1.1.1.1 # DNS 你看着改
VPN_DNS_SRV2=1.0.0.1 # DNS 你看着改
VPN_CLIENT_NAME=001 # 第一个客户端的名称
```

同目录下创建个 `vi docker-compose.yaml` 写入以下内容
```yaml
services:
  IKEv2:
    image: hwdsl2/ipsec-vpn-server
    container_name: IKEv2
    networks:
      IKEv2:
        ipv4_address: 172.18.0.2
    volumes:
      - ./ikev2-vpn-data:/etc/ipsec.d
      - /lib/modules:/lib/modules:ro
    privileged: true
    restart: always
    env_file:
      - ./.env

networks:
  IKEv2:
    external: true
```

最后 `docker compose up -d` 就完成了
服务端的cli你自己参考下
```sh
# 添加一个客户端（使用默认选项）
docker exec -it IKEv2 ikev2.sh --addclient [client name]
# 导出一个已有的客户端的配置
docker exec -it IKEv2 ikev2.sh --exportclient [client name]
# 列出已有的客户端
docker exec -it IKEv2 ikev2.sh --listclients
# 显示使用信息
docker exec -it IKEv2 ikev2.sh -h
```

`ikev2-vpn-data`目录里有客户端的配置文件
`p12` 是PC端
`mobileconfig` 是苹果端
`sswan` 是安卓端
导出到相应设备用就行了

Ocean-TXT, 2024-10-29:
> > > > 2022年就有人提出，现在都2024年了，希望能加上就好，因为这个ipsec不需要下载APP，iPhone手机自带客户端，非常好用。
> > > 
> > > 
> > > https://github.com/hwdsl2/docker-ipsec-vpn-server 用docker整吧
> > 
> > 
> > 因为是小白按步骤做了确链接不上，请问有视频教程吗
> 
> 那里已经有非常详细的中文文档了…… 我就给你简单说下 能不能行就看你自己了 还不明白就好好翻阅文档 当成是一个挑战来完成它 先创建个docker网络 专门给VPN用 `docker network create --driver bridge --subnet 172.18.0.0/30 --gateway 172.18.0.1 IKEv2`
> 
> 然后看看这个新建的网口ID叫啥 你会看到类似内容 之后在op的添加一个 就叫 允许转发到 允许来自的转发 找找 注意你的网口ID是不同的 参考前几句`docker network ls``48f0a288bc0c IKEv2``防火墙区域``区域``IKEv2``全接受``wan``lan``lan``高级设置``涵盖的设备``br-48f0a288bc0c`
> 
> 端口转发 把 都转发到 区域的`UDP 500``UDP 4500``IKEv2``172.18.1.2`
> 
> 创建 注意你的网口ID是不同的 参考前几句`NAT规则``任何``lan``任意``任意``MASQUERADE``高级设置``额外参数``-i br-48f0a288bc0c`
> 
> 创建个环境变量文件 写入以下内容`vi .env`
> 
> ```
> VPN_IKEV2_ONLY=yes # 只使用IKEv2模式
> VPN_DNS_NAME=你域名
> VPN_DNS_SRV1=1.1.1.1 # DNS 你看着改
> VPN_DNS_SRV2=1.0.0.1 # DNS 你看着改
> VPN_CLIENT_NAME=001 # 第一个客户端的名称
> ```
> 
> 同目录下创建个 写入以下内容`vi docker-compose.yaml`
> 
> ```yaml
> services:
>   IKEv2:
>     image: hwdsl2/ipsec-vpn-server
>     container_name: IKEv2
>     networks:
>       IKEv2:
>         ipv4_address: 172.18.0.2
>     volumes:
>       - ./ikev2-vpn-data:/etc/ipsec.d
>       - /lib/modules:/lib/modules:ro
>     privileged: true
>     restart: always
>     env_file:
>       - ./.env
> 
> networks:
>   IKEv2:
>     external: true
> ```
> 
> 最后 就完成了 服务端的cli你自己参考下`docker compose up -d`
> 
> ```shell
> # 添加一个客户端（使用默认选项）
> docker exec -it IKEv2 ikev2.sh --addclient [client name]
> # 导出一个已有的客户端的配置
> docker exec -it IKEv2 ikev2.sh --exportclient [client name]
> # 列出已有的客户端
> docker exec -it IKEv2 ikev2.sh --listclients
> # 显示使用信息
> docker exec -it IKEv2 ikev2.sh -h
> ```
> 
> `ikev2-vpn-data`目录里有客户端的配置文件 是PC端 是苹果端 是安卓端 导出到相应设备用就行了`p12``mobileconfig``sswan`

非常感谢你的回复，情况是这样的，我用的里面的的一键脚本
docker run \
    --name ipsec-vpn-server \
    --restart=always \
    -v ikev2-vpn-data:/etc/ipsec.d \
    -v /lib/modules:/lib/modules:ro \
    -p 500:500/udp \
    -p 4500:4500/udp \
    -d --privileged \
    hwdsl2/ipsec-vpn-server
安装好后用
docker logs ipsec-vpn-server
这个命令查看到了
Server IP: 198.18.8.6
IPsec PSK: GstNVk59o6MFkV9m6GsJ
Username: vpnuser
Password: NPRqYhWUkp8h6Dk3

但是198.18.8.6并非我的公网ip,然后我就不知道如何操作了还望知道一下或者如何能联系上你，有TG吗，还请帮忙指导一下，折腾几个月了。

closed 2025-09-13 by w9315273

kc-bao, 2025-10-08:
此问题已在 **酷友社** 上被提及。那里可能有相关详细信息：

https://www.koolcenter.com/t/topic/3361/1

w9315273, 2025-11-09:
> 可以自己下载插件安装 我一直在用

不如用docker.