istoreos/istoreos #276: 可以获取IPv6地址但不能访问IPv6网站

sumiles, 2022-07-02:
物理机安装，能获取到IPv6地址，但是访问不了IPv6网站
版本：OpenWrt 21.02.2 2022062120 / LuCI openwrt-21.02-cn branch git-22.165.12986-fdfc730
[image]
LAN口设置：
[image]
本机地址获取情况：
[image]
IPv6测试情况：
[image]

sumiles, 2022-07-02:
已经解决，passwll插件dns设置错误导致

closed 2022-07-02 by sumiles

mcmichaelchan, 2023-04-18:
> 

> 已经解决，passwll插件dns设置错误导致

@sumiles 请问你是如何修改passwall的设置从而解决问题的？