istoreos/istoreos #283: 重启死机；更新最新固件（5.4.179）无法安装软件提示内存不足

nisumlion, 2022-07-04:
重启死机无法进入系统
硬件：J1900 2G 128G 
X86 物理机

新固件无法安装更新软件（5.4.179）
安装和更新软件提示：
ff56a7786172: Download complete
3006eb83330b: Verifying Checksum
3006eb83330b: Download complete
failed to register layer: Error processing tar file(exit status 1): write /usr/local/lib/python3.9/site-packages/botocore/data/ec2/2016-09-15/service-2.json: no space left on device
98df0ecd742b3da92612f4fd76723ff6f8bfc2f8ce55b561cb11ee0e23a70589

jjm2473, 2022-07-04:
执行 blkid 看看

nisumlion, 2022-07-05:
> 

> 执行 blkid 看看

非IT专业不是很懂，也才玩路由器没多久。全靠保姆教程过来的

nisumlion, 2022-07-05:
> > 
> 
> > 执行 blkid 看看
> 
> 非IT专业不是很懂，也才玩路由器没多久。全靠保姆教程过来的

root@iStoreOS:~# blkid
/dev/sda3: LABEL="etc" UUID="19ef10cd-2089-4e67-a422-873f2bc65e5d" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="70d0a6fd-03"
/dev/sda1: LABEL="kernel" UUID="84173db5-fa99-e35a-95c6-28613cc79ea9" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="70d0a6fd-01"
/dev/sda2: TYPE="squashfs" PARTUUID="70d0a6fd-02"
/dev/sda4: UUID="01409401-cd2f-49be-abfc-a52e1e9ed84a" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="70d0a6fd-04"
root@iStoreOS:~# ^C
root@iStoreOS:~#

jjm2473, 2022-07-05:
你这个是docker拉取镜像的日志，看下docker的数据目录设置的是什么，还有执行`mount`和`df -h`两个命令看看

nisumlion, 2022-07-06:
麻烦大神帮忙看看，是在没看懂  哈哈 

root@iStoreOS:~# blkid
/dev/sda3: LABEL="etc" UUID="19ef10cd-2089-4e67-a422-873f2bc65e5d" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="70d0a6fd-03"
/dev/sda1: LABEL="kernel" UUID="84173db5-fa99-e35a-95c6-28613cc79ea9" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="70d0a6fd-01"
/dev/sda2: TYPE="squashfs" PARTUUID="70d0a6fd-02"
/dev/sda4: UUID="01409401-cd2f-49be-abfc-a52e1e9ed84a" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="70d0a6fd-04"
root@iStoreOS:~# mount
/dev/sda2 on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda3 on /overlay type ext4 (rw,relatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /mnt type tmpfs (rw,relatime,size=4096k)
/dev/sda1 on /boot type ext4 (rw,noatime)
/dev/sda1 on /boot type ext4 (rw,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
/dev/sda4 on /mnt/data_sda4 type ext4 (rw,relatime)
cgroup on /sys/fs/cgroup type tmpfs (rw,relatime,mode=755)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu type cgroup (rw,relatime,cpu)
cgroup on /sys/fs/cgroup/cpuacct type cgroup (rw,relatime,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,relatime,net_cls)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,relatime,perf_event)
cgroup on /sys/fs/cgroup/net_prio type cgroup (rw,relatime,net_prio)
cgroup on /sys/fs/cgroup/hugetlb type cgroup (rw,relatime,hugetlb)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,relatime,rdma)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)
/dev/sda3 on /overlay/upper/opt/docker type ext4 (rw,relatime)
overlay on /overlay/upper/opt/docker/overlay2/5e83155ddc1b3ea45d3cc0191d42f229950eca64b1dcd2092f3c27071ed926cd/merged type overlay (rw,relatime,lowerdir=/overlay/upper/opt/docker/overlay2/l/6JDX4N2LAG5F2JAH3D3M2TTCG4:/overlay/upper/opt/docker/overlay2/l/NVSXYZQXSNUJWZRMQK7XE2KMQS:/overlay/upper/opt/docker/overlay2/l/XR3OINBLNPW7ADDF2TD46ULDYA:/overlay/upper/opt/docker/overlay2/l/6FD62HK745OI3ASBLBCTUY3F2G:/overlay/upper/opt/docker/overlay2/l/BQRNNLLWXMULK5UD2UNBTDXOE5:/overlay/upper/opt/docker/overlay2/l/F3YPEMIGSOV2EIQL3ODT5U2BXQ:/overlay/upper/opt/docker/overlay2/l/3VIGB6Q6AXJZUDSU6TTRLPHZMH:/overlay/upper/opt/docker/overlay2/l/DGUD7N6UCBRYNIJELP3IRLMTSW:/overlay/upper/opt/docker/overlay2/l/NOXGJGSQ2PDRM26PCYNJT5TNTO:/overlay/upper/opt/docker/overlay2/l/NU4HCR6IJAAYWPLZY6DVGTCY7O:/overlay/upper/opt/docker/overlay2/l/VMX2DUJAB53QX4YCJVLRRZPMYV:/overlay/upper/opt/docker/overlay2/l/GQDEYE7PNVJDTKBMAUSJBXZ5DM:/overlay/upper/opt/docker/overlay2/l/I62UF6DB4BZBR4MLSBCJ6AT3Y2:/overlay/upper/opt/docker/overlay2/l/MREMMDNT7533QSCA5GCHM7N6E7:/overlay/upper/opt/docker/overlay2/l/BNQM4K6S4KUNE6KMJK7H5VY5YL:/overlay/upper/opt/docker/overlay2/l/BG74JJMZO236CFWL4H77CQI2MN:/overlay/upper/opt/docker/overlay2/l/7XKJRIFXPSMEE4NL6IDYJCIDH7:/overlay/upper/opt/docker/overlay2/l/PQ26CRBI3F2BCZF6ZAKL4NW4JQ:/overlay/upper/opt/docker/overlay2/l/ITU7SNGV2R3F5ILXR73TS3QGP6:/overlay/upper/opt/docker/overlay2/l/K4FYVVBABDIQ7EVDLB5N4WEVQG:/overlay/upper/opt/docker/overlay2/l/ZGKZOKNRARGB7PBYNV24COVJPI:/overlay/upper/opt/docker/overlay2/l/3YNVMNAUXNTLWF3NYRX2W6DMLI:/overlay/upper/opt/docker/overlay2/l/BC6D5UZVG7I237ZD44LQ5ISZJ6:/overlay/upper/opt/docker/overlay2/l/TCHK26XBBPII7MIUHFO2I6KEWK:/overlay/upper/opt/docker/overlay2/l/ITXM7FNQQQLSTOM7GCGNAYSVYV:/overlay/upper/opt/docker/overlay2/l/6CECOQFWT7APOV2SIED7DDUSYU:/overlay/upper/opt/docker/overlay2/l/7NT3URM7BEMH66OWDCB6XWVEBI:/overlay/upper/opt/docker/overlay2/l/KDHT7S6HFH6BVWFNPHJLSYM6ZP:/overlay/upper/opt/docker/overlay2/l/WPLGOAXOJG3B6DWYYDYF64CAUU,upperdir=/overlay/upper/opt/docker/overlay2/5e83155ddc1b3ea45d3cc0191d42f229950eca64b1dcd2092f3c27071ed926cd/diff,workdir=/overlay/upper/opt/docker/overlay2/5e83155ddc1b3ea45d3cc0191d42f229950eca64b1dcd2092f3c27071ed926cd/work,xino=off)
nsfs on /tmp/run/docker/netns/default type nsfs (rw)
root@iStoreOS:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               112.5M    112.5M         0 100% /rom
tmpfs                   943.4M      3.5M    939.9M   0% /tmp
/dev/sda3                 1.9G      1.5G    395.1M  80% /overlay
overlayfs:/overlay        1.9G      1.5G    395.1M  80% /
tmpfs                     4.0M         0      4.0M   0% /mnt
/dev/sda1               126.0M      5.1M    118.3M   4% /boot
/dev/sda1               126.0M      5.1M    118.3M   4% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/sda4               114.5G     60.0M    108.6G   0% /mnt/data_sda4
cgroup                  943.4M         0    943.4M   0% /sys/fs/cgroup
/dev/sda3                 1.9G      1.5G    395.1M  80% /overlay/upper/opt/docker
overlay                   1.9G      1.5G    395.1M  80% /overlay/upper/opt/docker/overlay2/5e83155ddc1b3ea45d3cc0191d42f229950eca64b1dcd2092f3c27071ed926cd/merged
root@iStoreOS:~#

jjm2473, 2022-07-07:
```/dev/sda3 1.9G 1.5G 395.1M 80% /overlay/upper/opt/docker```，说明你的docker数据所在分区只有400M空间了，如果你拉取的docker镜像超过这个大小就不够空间了啊。我看你的sda4分区比较大，首页那边的docker向导把数据迁移到这个分区，就有足够空间了。

nisumlion, 2022-07-07:
把数据迁移一下好像可以了 ，谢谢宝 

爱你~

closed 2022-07-08 by jannson