istoreos/istoreos #296: 关于群晖upnp无法端口映射问题，希望新系统继承lede上的DMZ插件的问题。

ghc918, 2022-07-11:
已经安装最新版的0707 X86版本的系统，群晖使用upnp，端口转发等无法转发、映射出去，其余功能暂时没有发现问题，希望大佬们可以和Lede上在酷软中有个DMZ小插件来解决直接DMZ出去的。虽然知道DMZ有很大很大的安全风险，但是此类办法对于使用软路由系统的半小白来说非常的实用，放在软件中心里面也让使用者明白DMZ功能的风险。
   关于DMZ，各位大佬请轻喷、轻喷，我知道这个风险，但是难免有些会出现这种问题

jannson, 2022-07-11:
不是旁路由吧？
UPNP服务打开了么？
试试安装一个易有云，看看能否把 UPNP 转发出去。如果不行可以加群，我们可以确认下。UPNP 我们之前确认肯定是可以的，并且我们开发的易有云软件也依赖这个特性。。。

ghc918, 2022-07-12:
> 不是可以由我们方便打开 一个吧？开发的易有云软件也依赖这个特性。。。

下次空了再刷一次试一试易有云的问题，不是旁路由，物理机直接刷的，物理机是联想的M93p，CPU用的i3-4170我在电报群里也发了各种截图的，确实没办法了。不过我昨天刷回lede的时候我发现了个问题，好像istoreos的5000端口被占用了，群晖写upnp的时候每次那个端口都会被强制换另外一个，群晖上强制应用也不行。就彻底放弃了，等老大佬们继续完善。

jannson, 2022-07-18:
istoreos 5000 端口被谁占用了呢？能否说明下，我们看看能否兼容下群晖。群晖用户还是多。

ghc918, 2022-07-19:
> istoreos 5000 端口被谁占用了呢？能否说明下，我们看看能否兼容下群晖。群晖用户还是多。

这个被谁占用了我就真不知道了。确实每次群晖自动映射的时候，5000端口就被自己更改到五位数的端口了，有空我再刷一个试试。

bcseputetto, 2022-07-26:
> > istoreos 5000 端口被谁占用了呢？能否说明下，我们看看能否兼容下群晖。群晖用户还是多。
> 
> 这个被谁占用了我就真不知道了。确实每次群晖自动映射的时候，5000端口就被自己更改到五位数的端口了，有空我再刷一个试试。

其实是miniupnp自己占用了5000端口，你在upnp页面不是看到有个端口，写的是5000，自己改成4000就行了

jjm2473, 2022-07-27:
dmz本质就是端口转发，只是端口范围是1-65535，直接在防火墙就能配置出这种效果，没必要弄个插件

ghc918, 2022-07-27:
> dmz本质就是端口转发，只是端口范围是1-65535，直接在防火墙就能配置出这种效果，没必要弄个插件

你当我不知道？全范围设置转发什么  群晖改端口全部尝试，失败，我去恩山找了个其他版本的就单独设置端口转发就行，这明显要么要么固件，要么硬件。反正我没找到原因

jjm2473, 2022-07-27:
那跟dmz有啥关系呢？

jjm2473, 2022-07-27:
5000端口是miniupnp占用的，不过并不影响端口转发，因为端口转发是防火墙的功能，端口转发的优先级高于本机的端口监听

closed 2022-09-22 by jannson