istoreos/istoreos #300: Docker不能拉取镜像 错误代码400

KeplerSSSSS, 2022-07-13:
Docker不能拉取镜像 错误代码400  不知道是哪里出了问题

w9315273, 2022-07-14:
400分两种 bad request和invalid hostname

Docker——配置——注册表镜像——检查地址
iStoreOS默认值是 https://docker.mirrors.ustc.edu.cn/

检查镜像名称格式是否输入正确 例如 lisaac/luci:latest 作者/镜像名称:版本

KeplerSSSSS, 2022-07-15:
我的400是bad request  镜像地址用的就是默认的

镜像名称是在Docker官网复制的

closed 2022-07-18 by jannson

dpwfor, 2024-04-15:
failed to mknod("/etc/apache2/conf-enabled/charset.conf", S_IFCHR, 0): operation not permitted
 拉取后 错误提示

dpwfor, 2024-04-15:
> 400分两种 bad request和invalid hostname
> 
> Docker——配置——注册表镜像——检查地址 iStoreOS默认值是 https://docker.mirrors.ustc.edu.cn/
> 
> 检查镜像名称格式是否输入正确 例如 lisaac/luci:latest 作者/镜像名称:版本

 failed to mknod("/etc/apache2/conf-enabled/charset.conf", S_IFCHR, 0): operation not permitted
拉取后错误提示