istoreos/istoreos #302: X86旁路由模式问题

My-By, 2022-07-13:
X86：舍弃WAN口后，LAN口全桥接，手机连接到主路由的WiFi并更改ip，网关指向软路由就无法上网。
（经过几天测试）只有开了OPClash后所有网络正常，启动后翻不翻得了墙都能正常上网；Passwall也不行，其他翻墙软件不常用没尝试。
网线设备指向软路由的都可以正常上网。

w9315273, 2022-07-14:
在WIFI路由器是检查是否开启了类似 “禁止无线用户互相访问”
尝试在主路由上将网关设为旁路由

“只有开了OpenClash后所有网络正常” 
检查旁路由的网关是否设为了主路由 DNS是否填写正确

My-By, 2022-07-15:
> 

网关和NDS的设置都没有问题，因为主路由需要分配ip所以没办法把主路由设置成旁路由，因为以前别的固件设置方法都是一样的，所以ISOS不能用我才觉得奇怪，未把wifi设备指向旁路由之前，都是能正常wifi上网的。

w9315273, 2022-07-15:
> 

我是说
主路由的网关设为旁路由
旁路由的网关设为主路由
这不影响主路由的DHCP分配
你现在不是在客户端上手动把网关指定到旁路由吗

My-By, 2022-07-15:
> > 
> 
> 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗

互相指向吗，这个倒是没试过，我试试。

w9315273, 2022-07-15:
> > 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗
> 
> 互相指向吗，这个倒是没试过，我试试。

然后客户端全自动就好了 不用手动填写

My-By, 2022-07-15:
> > > 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗
> > 
> > 
> > 互相指向吗，这个倒是没试过，我试试。
> 
> 然后客户端全自动就好了 不用手动填写

刚试了一下，双方网关互指，关闭Openclash后，wifi设备依旧无法上网，但是网线设备依旧正常。
不过刚发现，wifi连接的设备会自动出现在isos的路由表里，没指向的也会出现，以前就没遇到过。只有指向的才会出现在路由表里，不知道这个有没有什么相关。

w9315273, 2022-07-15:
> > > > 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗
> > > 
> > > 
> > > 互相指向吗，这个倒是没试过，我试试。
> > 
> > 
> > 然后客户端全自动就好了 不用手动填写
> 
> 刚试了一下，双方网关互指，关闭Openclash后，wifi设备依旧无法上网，但是网线设备依旧正常。 不过刚发现，wifi连接的设备会自动出现在isos的路由表里，没指向的也会出现，以前就没遇到过。只有指向的才会出现在路由表里，不知道这个有没有什么相关。

没关系
把主路由获取到的运营商DNS 填写到旁路由里

My-By, 2022-07-15:
> > > > > 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗
> > > > 
> > > > 
> > > > 互相指向吗，这个倒是没试过，我试试。
> > > 
> > > 
> > > 然后客户端全自动就好了 不用手动填写
> > 
> > 
> > 刚试了一下，双方网关互指，关闭Openclash后，wifi设备依旧无法上网，但是网线设备依旧正常。 不过刚发现，wifi连接的设备会自动出现在isos的路由表里，没指向的也会出现，以前就没遇到过。只有指向的才会出现在路由表里，不知道这个有没有什么相关。
> 
> 没关系 把主路由获取到的运营商DNS 填写到旁路由里

这个我倒是有设置。

w9315273, 2022-07-15:
> > > > > > 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗
> > > > > 
> > > > > 
> > > > > 互相指向吗，这个倒是没试过，我试试。
> > > > 
> > > > 
> > > > 然后客户端全自动就好了 不用手动填写
> > > 
> > > 
> > > 刚试了一下，双方网关互指，关闭Openclash后，wifi设备依旧无法上网，但是网线设备依旧正常。 不过刚发现，wifi连接的设备会自动出现在isos的路由表里，没指向的也会出现，以前就没遇到过。只有指向的才会出现在路由表里，不知道这个有没有什么相关。
> > 
> > 
> > 没关系 把主路由获取到的运营商DNS 填写到旁路由里
> 
> 这个我倒是有设置。

那我不会了

My-By, 2022-07-15:
> > > > > > > 我是说 主路由的网关设为旁路由 旁路由的网关设为主路由 这不影响主路由的DHCP分配 你现在不是在客户端上手动把网关指定到旁路由吗
> > > > > > 
> > > > > > 
> > > > > > 互相指向吗，这个倒是没试过，我试试。
> > > > > 
> > > > > 
> > > > > 然后客户端全自动就好了 不用手动填写
> > > > 
> > > > 
> > > > 刚试了一下，双方网关互指，关闭Openclash后，wifi设备依旧无法上网，但是网线设备依旧正常。 不过刚发现，wifi连接的设备会自动出现在isos的路由表里，没指向的也会出现，以前就没遇到过。只有指向的才会出现在路由表里，不知道这个有没有什么相关。
> > > 
> > > 
> > > 没关系 把主路由获取到的运营商DNS 填写到旁路由里
> > 
> > 
> > 这个我倒是有设置。
> 
> 那我不会了

我估计是固件有什么遗漏的，因为在刚安装好，然后只关了WAN口，桥接了LAN口，网关指向主路由，NDS，关闭DHCP的情况下都没办法wifi上网，固件也装过不少了，也摸索过，每次安装新固件，网口设置都是一样的，就ISOS没办法旁路由上网，应该还需要改进吧。

w9315273, 2022-07-15:
> 我估计是固件有什么遗漏的，因为在刚安装好，然后只关了WAN口，桥接了LAN口，网关指向主路由，NDS，关闭DHCP的情况下都没办法wifi上网，固件也装过不少了，也摸索过，每次安装新固件，网口设置都是一样的，就ISOS没办法旁路由上网，应该还需要改进吧。

固件没问题 我用旁路由模式就没问任何问题
你的主路由是什么 我的是AX88U

My-By, 2022-07-15:
> 固件没问题 我用旁路由模式就没问任何问题 你的主路由是什么 我的是AX88U

小米AX9000

My-By, 2022-07-15:
> 小米AX9000

固件没问题吗，你用的什么版本的，0707的吗，软路由是x86吗，如果是的话，难道是兼容问题？

w9315273, 2022-07-15:
x86 0707 我觉得你的主路由设置有问题

My-By, 2022-07-15:
> x86 0707 我觉得你的主路由设置有问题

既然不是所有人都有这个问题，那估计就是我的问题，我研究一下主路由。

My-By, 2022-07-15:
> x86 0707 我觉得你的主路由设置有问题

很奇妙，主路由设置能调的都调了，依旧如此，apMesh连接到主路由的Mesh路由的wifi设备指向旁路由不开OPENclash却可以正常上网，应该是主路由问题，或者ISOS系统与AX9000不兼容，以前用别的软路由系统没出现过这种情况。
不过还是谢谢你，虽然没解决，至少知道是主路由的问题了。

My-By, 2022-07-15:
> x86 0707 我觉得你的主路由设置有问题

总算是解决了，小米路由做主路由的话，就得开启防火墙里LAN→WAN的ip动态伪装，然后就好了，狗贼，让我研究那么久，现在所有功能正常了。

closed 2022-07-15 by My-By