istoreos/istoreos #306: ipk install issue

jasonyy2018, 2022-07-16:
...
Error loading shared library libwolfssl.so.5.2.0.99a5b54a: No such file or directory (needed by /usr/bin/curl)
Error relocating /usr/bin/curl: curl_easy_header: symbol not found
Error relocating /usr/bin/curl: curl_easy_nextheader: symbol not found
Collected errors:
 * opkg_download: Failed to download https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/21.02.2/packages/aarch64_generic/packages/libcap-bin_2.43-1_aarch64_generic.ipk, curl returned 127.
 * opkg_install_pkg: Failed to download libcap-bin. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci-app-openclash.

离线安装失败,错误码: 255

GreenTeodoro839, 2022-07-16:
opkg update再试试

jannson, 2022-07-18:
原因是，你更新了其它的插件，覆盖了系统的 curl 文件。导致更新失败。。。
可以到群里找到一个 reset-rom-xxx.run 恢复系统重要文件，或者重新安装固件。

closed 2022-07-18 by jannson