istoreos/istoreos #309: 希望能增加对友善 NanoPi R2C 的支持

krkrchan, 2022-07-16:
NanoPi R2 系列有三款产品：R2S、R2C、R2S Plus。

据官方介绍 R2S 和 R2C 的不同是，R2C 使用了国产 YT8521S 网卡，其他都一致。

因此，应该在在固件里额外增加一个 YT8521S 网卡驱动，就能实现对 R2C 的支持了。

希望开发者能支持一波，谢谢！谢谢！谢谢！

附官方参数比较：https://img.alicdn.com/imgextra/i2/73291228/O1CN01lu2rYZ1KwT5zp8qAG_!!73291228.jpg

jjm2473, 2022-12-16:
已添加相关驱动

closed 2022-12-16 by jjm2473

krkrchan, 2022-12-19:
您好，我下载了最新的R2S固件包，但是WAN口还是无法工作🧐指示灯也不亮

jjm2473, 2022-12-20:
@krkrchan 开机以后执行 `dmesg > /tmp/dmesg.log` ，然后将 /tmp/dmesg.log 文件发出来看看

krkrchan, 2022-12-29:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.188 (build@buildserver) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 2022120817)) #0 SMP PREEMPT Fri Dec 16 06:01:32 2022
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s47832 r8192 d29992 u86016
[    0.000000] pcpu-alloc: s47832 r8192 d29992 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1004576K/1046528K available (7806K kernel code, 572K rwdata, 2396K rodata, 1664K init, 713K bss, 33760K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000009] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001569] Console: colour dummy device 80x25
[    0.002029] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002961] pid_max: default: 32768 minimum: 301
[    0.003638] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004318] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007335] ASID allocator initialised with 32768 entries
[    0.007986] rcu: Hierarchical SRCU implementation.
[    0.009647] smp: Bringing up secondary CPUs ...
[    0.010843] Detected VIPT I-cache on CPU1
[    0.010925] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011745] Detected VIPT I-cache on CPU2
[    0.011806] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012609] Detected VIPT I-cache on CPU3
[    0.012670] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012797] smp: Brought up 1 node, 4 CPUs
[    0.016104] SMP: Total of 4 processors activated.
[    0.016537] CPU features: detected: 32-bit EL0 Support
[    0.017007] CPU features: detected: CRC32 instructions
[    0.025412] CPU: All CPU(s) started at EL2
[    0.025876] alternatives: patching kernel code
[    0.039983] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.040889] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.042186] pinctrl core: initialized pinctrl subsystem
[    0.043919] NET: Registered protocol family 16
[    0.046782] DMA: preallocated 256 KiB pool for atomic allocations
[    0.048549] cpuidle: using governor menu
[    0.049169] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
[    0.050063] Serial: AMBA PL011 UART driver
[    0.079257] HugeTLB registered 1.00 GiB page size, pre-allocated 0 pages
[    0.079883] HugeTLB registered 32.0 MiB page size, pre-allocated 0 pages
[... 455 lines cut ...]
[   22.765740] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   23.452044] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   30.449544] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   30.450293] NFSD: Using legacy client tracking operations.
[   30.450779] NFSD: starting 10-second grace period (net f0000040)
[   72.381992] br-lan: port 1(eth1) entered disabled state
[   72.384972] r8152 4-1:1.0 eth1: carrier off
[   97.118997] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   97.119429] r8152 4-1:1.0 eth1: carrier on
[   97.120249] br-lan: port 1(eth1) entered blocking state
[   97.120268] br-lan: port 1(eth1) entered forwarding state
[... cut ...]
[  116.191366] r8152 4-1:1.0 eth1: carrier on
[  116.192011] br-lan: port 1(eth1) entered blocking state
[  116.192027] br-lan: port 1(eth1) entered forwarding state
[  135.005508] br-lan: port 1(eth1) entered disabled state
[  135.011470] r8152 4-1:1.0 eth1: carrier off
[  137.950987] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[  137.951362] r8152 4-1:1.0 eth1: carrier on
[  137.952012] br-lan: port 1(eth1) entered blocking state
[  137.952026] br-lan: port 1(eth1) entered forwarding state

```

krkrchan, 2022-12-30:
@jjm2473 麻烦看一下上面的日志哦，谢谢。

jjm2473, 2023-01-03:
看不出来，也没有r2c可以测试

daiwei43211, 2023-01-03:
希望同样支持对于R2C的固件。谢谢群主

krkrchan, 2023-01-03:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.188 (build@buildserver) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 2022120817)) #0 SMP PREEMPT Fri Dec 16 06:01:32 2022
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s47832 r8192 d29992 u86016
[    0.000000] pcpu-alloc: s47832 r8192 d29992 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1004576K/1046528K available (7806K kernel code, 572K rwdata, 2396K rodata, 1664K init, 713K bss, 33760K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001590] Console: colour dummy device 80x25
[    0.002046] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002977] pid_max: default: 32768 minimum: 301
[    0.003672] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004353] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007368] ASID allocator initialised with 32768 entries
[    0.008024] rcu: Hierarchical SRCU implementation.
[    0.009718] smp: Bringing up secondary CPUs ...
[    0.010904] Detected VIPT I-cache on CPU1
[    0.010989] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011811] Detected VIPT I-cache on CPU2
[    0.011874] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012691] Detected VIPT I-cache on CPU3
[    0.012755] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012882] smp: Brought up 1 node, 4 CPUs
[    0.016183] SMP: Total of 4 processors activated.
[    0.016616] CPU features: detected: 32-bit EL0 Support
[    0.017085] CPU features: detected: CRC32 instructions
[    0.025499] CPU: All CPU(s) started at EL2
[    0.025967] alternatives: patching kernel code
[    0.040103] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.041008] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.042298] pinctrl core: initialized pinctrl subsystem
[    0.044022] NET: Registered protocol family 16
[    0.046878] DMA: preallocated 256 KiB pool for atomic allocations
[    0.048642] cpuidle: using governor menu
[    0.049255] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
[    0.050151] Serial: AMBA PL011 UART driver
[    0.079338] HugeTLB registered 1.00 GiB page size, pre-allocated 0 pages
[    0.079965] HugeTLB registered 32.0 MiB page size, pre-allocated 0 pages
[... 449 lines cut ...]
[   36.213171] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[   36.323337] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
[   36.323939] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   36.324751] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_open: Hw setup failed
[   37.147375] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   37.731067] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   43.175822] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   43.176582] NFSD: Using legacy client tracking operations.
[... cut ...]
[   52.713473] overlayfs: upperdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   52.713487] overlayfs: workdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   52.713494] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.
[   52.713500] overlayfs: lowerdir is in-use as upperdir/workdir of another mount, accessing files from both mounts will result in undefined behavior.

```

krkrchan, 2023-01-03:
@jjm2473 作者您好，您看一下36.213171秒开始的信息，好像调用的 RTL8211E 驱动欸。R2C是 YT8521S 驱动。
```
[   36.213171] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[   36.323337] rk_gmac-dwmac ff540000.ethernet: Failed to reset the dma
[   36.323939] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   36.324751] rk_gmac-dwmac ff540000.ethernet eth0: stmmac_open: Hw setup failed
```

jjm2473, 2023-01-30:
https://github.com/istoreos/istoreos/issues/566