istoreos/istoreos #312: 静态路由设置后能 ping 通但打不开网页

nuintun, 2022-07-18:
[image]

如图，R2S 安装 iStoreOS，下面接了 XDR3040 当 AP， XDR3040 的 LAN 口连接蒲公英 G5 做旁路，其它设备 WIFI 接入

G5 WAN 地址： 192.168.50.3
G5 LAN 地址： 10.168.2.1
G5 旁路路由： 192.168.31.0/24

静态路由设置
10.168.2.0/24      255.255.255.0    192.168.50.3    LAN
192.168.31.0/24    255.255.255.0    192.168.50.3    LAN

测试能 ping 通 192.168.31.0/24 下的异地设备和 10.168.2.0/24 下的本地设备，但是网页无法访问。

192.168.31.1 为异地路由管理地址，以前华硕路由同样静态路由设置可以互访！

请问这个是什么原因照成的？怎么解决？😶

jannson, 2022-07-18:
很多路由器都有防火墙，很可能就是防火墙拦截了你的 80 或者 8443 的端口。你可以开启相关路由器的端口，让他能外网访问，就可以了。

closed 2022-07-18 by jannson

nuintun, 2022-07-18:
@jannson 蒲公英 G5 WAN 口转发是打开的，同样设置华硕路由静态路由是没问题的！