istoreos/istoreos #313: 希望可以加入usb网卡的驱动cm-811ac与CF-812AC网卡的驱动

xiaoai126, 2022-07-18:
希望可以加入usb网卡的驱动cm-811ac与CF-812AC网卡的驱动 非常感谢

LLP2333, 2022-10-15:
我的设备是R2S
我也希望加入usb网卡的驱动comfast家的811ac，这个仓库可能有用https://github.com/aircrack-ng/rtl8812au
可以在淘宝购买到
【淘宝】https://m.tb.cn/h.U3JFlKz?tk=ilzG2z1ozKi CZ3457 「【MAC专用】COMFAST 811AC家用650M无线网卡台式机双频5g笔记本电脑主机wifi接收器USB随身发射器黑苹果Macos」
点击链接直接打开 或者 淘宝搜索直接打开
另外这个固件 https://github.com/DHDAXCW/NanoPi-R2S-rk3328是支持comfast的811ac的
关于R2S的官方wiki中有介绍  https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2S/zh
R2S的官方支持的usb网卡类型    https://wiki.friendlyelec.com/wiki/index.php/File:R2swrt%2Busbwifi-20210831.jpg

tangjw, 2023-03-05:
同求+1，友善官wiki的图 https://wiki.friendlyelec.com/wiki/images/4/4d/R2swrt%2Busbwifi-20210831.jpg
[image]

arnoliudaxia, 2024-07-21:
现在去后台装个RTL8821的软件包就行了