istoreos/istoreos #315: 手动无法安装插件

hmkdyl, 2022-07-20:
[image]

jannson, 2022-07-20:
你自己安装了相关插件，替换了系统的 openssl 或者 curl 或者 其它相关的包么？

或者考虑恢复出厂再试试。。。

hmkdyl, 2022-07-20:
没有安装任何自己的插件，这个系统怎么找不到恢复出厂设置的开关

ziguayungui, 2022-07-20:
你是在R2S中安装***_x86.run吗   你装的这个是x86的包，不是r2s的

hmkdyl, 2022-07-20:
原来问题出在这里啊，那哪里有适合R2S的包啊，我没看到KoolCenter上面有呢

ziguayungui, 2022-07-20:
去这个地址上找一下  https://github.com/AUK9527/Are-u-ok/tree/main/apps/all

hmkdyl, 2022-07-20:
> 去这个地址上找一下 https://github.com/AUK9527/Are-u-ok/tree/main/apps/all

好的，谢谢你

hmkdyl, 2022-07-20:
用这里的插件显示安装完成，可是在已安装插件里却找不到，搞不懂了
[image]

ziguayungui, 2022-07-20:
重启一下设备
图标不在已安装的插件里面，在服务等等菜单里面找。

hmkdyl, 2022-07-20:
是的，找到了，在服务列表里，已经能正常运行插件了，话说R2S的性能真的差了点，显示节点列表都要刷新半天，感谢帮忙！

closed 2022-07-24 by jjm2473