istoreos/istoreos #329: 无法安装多播插件

L1cardo, 2022-07-28:
固件为iStoreOS x86最新版

安装多播时出现以下画面，然后一直卡住，等了一个多小时还是这样。

如果强制关闭，那么系统环境必然出问题，比如更新软件包源就会提示curl有问题

[image]

whg112, 2022-07-31:
我的也是这样子不能安装。

jjm2473, 2022-08-08:
mwan3安装过程中会导致网络中断，日志就没继续刷了，刷新页面要是还出现对话框，说明还在安装，如果没出现对话框说明完成了，但可能失败

L1cardo, 2022-08-09:
@jjm2473 你可能没仔细看我说了什么

> 安装多播时出现以下画面，然后一直卡住，等了一个多小时还是这样。
如果强制关闭，那么系统环境必然出问题，比如更新软件包源就会提示curl有问题

jannson, 2022-08-25:
很可能因为某个原因，curl 被更新了，导致 Opkg 依赖的 curl 无法使用了。从而导致了后续的一堆一堆的问题！
我们得做一个系统检测功能，如果发现核心功能确实，就提示错误，并可以自动恢复系统核心的功能组件！

jjm2473, 2022-08-26:
istore日志中断的问题修复了。至于curl，没有截图不知道啥问题

L1cardo, 2022-08-27:
现在的情况是，安装或者卸载多播插件必出现在这个错误，点击确定或着取消都没用，反复跳
[image]

然后重启op，就出现这个错误，旧都救不活
[image]

jjm2473, 2022-08-29:
点取消没用？那肯定是浏览器问题。点取消以后，刷新页面应该会回到登陆页面。

jjm2473, 2022-08-29:
登陆异常肯定是因为文件损坏了，检查下是不是磁盘满了，或者文件系统故障