istoreos/istoreos #331: wireguard与设置成同一个防火墙区域后dhcp崩溃

le3pak, 2022-07-28:
opkg update之后安装wireguard，
设置服务端口与对端端口完毕后，设置wireguard端口与lan端口为一个防火墙区域后，dhcp不能分配ip，lan直接崩溃不能ping通，通过重置才能解决。
如果可以的话，请开发者修复此问题。

666dechaoge, 2022-12-11:
> opkg update之后安装wireguard， 设置服务端口与对端端口完毕后，设置wireguard端口与lan端口为一个防火墙区域后，dhcp不能分配ip，lan直接崩溃不能ping通，通过重置才能解决。 如果可以的话，请开发者修复此问题。

大佬你解决了吗，用不了wireguard难受

666dechaoge, 2022-12-14:
> opkg update之后安装wireguard， 设置服务端口与对端端口完毕后，设置wireguard端口与lan端口为一个防火墙区域后，dhcp不能分配ip，lan直接崩溃不能ping通，通过重置才能解决。 如果可以的话，请开发者修复此问题。

不在同一个防火墙能用吗（我R2S只添加wireguard接口，什么都还没配就崩了。。。

le3pak, 2022-12-16:
> > opkg update之后安装wireguard， 设置服务端口与对端端口完毕后，设置wireguard端口与lan端口为一个防火墙区域后，dhcp不能分配ip，lan直接崩溃不能ping通，通过重置才能解决。 如果可以的话，请开发者修复此问题。
> 
> 不在同一个防火墙能用吗（我R2S只添加wireguard接口，什么都还没配就崩了。。。

过去很久了，记得当时不在同一个防火墙，给wg端口新建的端口，也是崩溃不分配ip，多番试验无果。后面就把istoreOS弃了，用supes.top定制固件，目前使用稳定。

jjm2473, 2022-12-16:
下个版本会集成wireguard驱动，看看能不能解决

jjm2473, 2022-12-19:
升级最新固件，看看问题还存不存在，如果之前安装过wireguard驱动，先卸载掉再升级固件

closed 2023-02-18 by jjm2473