istoreos/istoreos #340: 配置IPTV

KittenCN, 2022-08-08:
从lede 2.34升级到istoreos，发现iptv不能看了，检查了下配置，有以下几个疑问：
1. 原先系统菜单下的进阶配置项目没有了
2. 接口中配置vlan51和vlan85的接口提示 未知错误（DEVICE_CLAIM_FAILED）

不知道是否是我操作的问题？

jjm2473, 2022-12-16:
你是在哪里配置vlan的，luci 21的网络配置分成了“接口”和“设备”，试试在“设备”页面增加VLAN虚拟接口

closed 2023-02-05 by KittenCN

tx5866, 2023-02-08:
我配置的时候也是出现
未知错误（DEVICE_CLAIM_FAILED）
这个怎么解决的？能不能截图看看？

KittenCN, 2023-02-09:
> 我配置的时候也是出现 未知错误（DEVICE_CLAIM_FAILED） 这个怎么解决的？能不能截图看看？

vlan85要配置dhcp模式，并且跃迁点放到20,然后再安装配置updxy就可以了，我现在是正常了,上海电信iptv

minwe, 2023-02-17:
@KittenCN 有没有详细一点的说明？谢谢！

我也是从 LEDE_X64_fw867 刷成 iStoreOS，从 LEDE 上复制过来的网络配置，也是报这个错。

jjm2473, 2023-02-18:
@minwe 不要去复制其他非官方openwrt的配置，尤其是lede，他们还用着非常老的系统组件，估计配置不会兼容的

jjm2473, 2023-02-18:
@KittenCN 所以你是从lede固件保留配置升级到istoreos？

KittenCN, 2023-02-20:
> @KittenCN 所以你是从lede固件保留配置升级到istoreos？

不是，原来的配置有问题，我全部重新设置了

minwe, 2023-02-27:
> > 我配置的时候也是出现 未知错误（DEVICE_CLAIM_FAILED） 这个怎么解决的？能不能截图看看？
> 
> vlan85要配置dhcp模式，并且跃迁点放到20,然后再安装配置updxy就可以了，我现在是正常了,上海电信iptv

直接在 VLAN 过滤里配置也可以，这周末搞了一下 OK 了。回头我整理个文档。

THYLTH, 2023-05-03:
> > > 我配置的时候也是出现 未知错误（DEVICE_CLAIM_FAILED） 这个怎么解决的？能不能截图看看？
> > 
> > 
> > vlan85要配置dhcp模式，并且跃迁点放到20,然后再安装配置updxy就可以了，我现在是正常了,上海电信iptv
> 
> 直接在 VLAN 过滤里配置也可以，这周末搞了一下 OK 了。回头我整理个文档。

有没有文档分享之类的。。我还是没搞定

minwe, 2023-05-22:
@THYLTH 

在 设备 -> br-lan 点配置，然后 VLAN 网桥过滤里面设置：

[image]

ITManCHINA, 2023-11-23:
> @THYLTH
> 
> 在 设备 -> br-lan 点配置，然后 VLAN 网桥过滤里面设置：
> 
> [image]
> 不要在接口里面设置。

@minwe 您好，我想请教一下您两个问题。
1.U和T标签对LAN口有何影响？
2.因为光猫中同时存在VLAN ID与组播VLAN两个值，请问是直接将光猫VLAN ID和组播VLAN两个值填入并启用即可还是有其它处理方式呢？
希望能得到您的回复。

jjm2473, 2023-11-24:
U表示出去的包不打标，也就是没有VLAN tag，T就是反过来。一个网口最多只能有一个Untag的VLAN，Tag的VLAN就没限制。

ITManCHINA, 2023-11-24:
> U表示出去的包不打标，也就是没有VLAN tag，T就是反过来。一个网口最多只能有一个Untag的VLAN，Tag的VLAN就没限制。

明白了，那么在iStoreOS中如何填入VLAN ID与组播VLAN的值呢？

minwe, 2023-11-27:
> > U表示出去的包不打标，也就是没有VLAN tag，T就是反过来。一个网口最多只能有一个Untag的VLAN，Tag的VLAN就没限制。
> 
> 明白了，那么在iStoreOS中如何填入VLAN ID与组播VLAN的值呢？

就是上面截图的最左侧的 VLAN ID 那里填，我的配置里 43 是给 IPTV 用的