istoreos/istoreos #349: 启用r4s

hakrei, 2022-08-18:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

jannson, 2022-08-25:
R4S 我们没精力支持。我们考虑支持 R5S ，目前 ARS2 已经支持！

closed 2022-08-25 by jannson

GreenTeodoro839, 2022-12-30:
> R4S 我们没精力支持。我们考虑支持 R5S ，目前 ARS2 已经支持！

看恩山上有帖子说gpu驱动的，lean版可以直接添加 
https://www.right.com.cn/forum/thread-4099256-1-1.html ，官方版打补丁后也可以 https://www.right.com.cn/forum/thread-5357320-1-1.html ，下个版本能不能加上gpu驱动试试，jellyfin就先让有需要的人自己折腾吧😀

enjoyO, 2023-06-17:
@jannson  请问线下r5c和R5s的固件已经支持GPU硬件解码了吗