istoreos/istoreos #357: Intel-i226-V网卡驱动添加

dnpl, 2022-08-26:
最新出的畅网第四版Intel-i226-V无法识别

blankvip, 2022-08-31:
我也使用了，无法识别，不知道什么时候能添加

Ethan1209, 2022-09-05:
mark一下，同样的问题

dnpl, 2022-09-06:
看了一下源码，有226的驱动更新，只是没有编译进去，我自己编译了一下，还没有测试能否识别，先关闭issue了

closed 2022-09-06 by dnpl

blankvip, 2022-09-10:
> 看了一下源码，有226的驱动更新，只是没有编译进去，我自己编译了一下，还没有测试能否识别，先关闭issue了

大佬，编译完可以用吗？

dnpl, 2022-09-10:
> > 看了一下源码，有226的驱动更新，只是没有编译进去，我自己编译了一下，还没有测试能否识别，先关闭issue了
> 
> 大佬，编译完可以用吗？

已经编译完成，自己测试使用目前没有发现问题，可以使用源码编译一下

blankvip, 2022-09-10:
> > > 看了一下源码，有226的驱动更新，只是没有编译进去，我自己编译了一下，还没有测试能否识别，先关闭issue了
> > 
> > 
> > 大佬，编译完可以用吗？
> 
> 已经编译完成，自己测试使用目前没有发现问题，可以使用源码编译一下

大佬，求共享固件，我不会编译，感谢。。。

dnpl, 2022-09-10:
> > > > 看了一下源码，有226的驱动更新，只是没有编译进去，我自己编译了一下，还没有测试能否识别，先关闭issue了
> > > 
> > > 
> > > 大佬，编译完可以用吗？
> > 
> > 
> > 已经编译完成，自己测试使用目前没有发现问题，可以使用源码编译一下
> 
> 大佬，求共享固件，我不会编译，感谢。。。

留下q，有空发给你一下

blankvip, 2022-09-10:
> > > > > 看了一下源码，有226的驱动更新，只是没有编译进去，我自己编译了一下，还没有测试能否识别，先关闭issue了
> > > > 
> > > > 
> > > > 大佬，编译完可以用吗？
> > > 
> > > 
> > > 已经编译完成，自己测试使用目前没有发现问题，可以使用源码编译一下
> > 
> > 
> > 大佬，求共享固件，我不会编译，感谢。。。
> 
> 留下q，有空发给你一下

q：626631052

shen-lan, 2023-01-10:
现在最新版的还有这个问题吗 ？