istoreos/istoreos #358: 希望r2s istore固件加入yt8521s驱动，现在我的r2cplus只能单臂路由，谢谢

waliwaliwa, 2022-08-26:
This repository is a mirror of our main repo at https://git.openwrt.org/openwrt/openwrt.git
It is for reference only and is not active for checkins or for reporting issues.

All issues should be reported at: https://bugs.openwrt.org

Please do not open any NEW issue here - we will be closing the Github issues
capability in the near future.

We will continue to accept Pull Requests here as described in the repo description.

Close this now and click here: https://bugs.openwrt.org

Thankyou for your co-operation.

jjm2473, 2022-12-16:
r2s固件加上yt8521s驱动就能支持r2c了吗？

jjm2473, 2022-12-16:
https://github.com/immortalwrt/immortalwrt/blob/openwrt-21.02/target/linux/rockchip/patches-5.4/600-net-phy-Add-driver-for-Motorcomm-YT85xx-PHYs.patch
https://github.com/immortalwrt/immortalwrt/blob/openwrt-21.02/target/linux/rockchip/files/include/linux/motorcomm_phy.h
https://github.com/immortalwrt/immortalwrt/blob/openwrt-21.02/target/linux/rockchip/files/drivers/net/phy/motorcomm.c

jjm2473, 2022-12-16:
yt8521s驱动加上了

closed 2022-12-16 by jjm2473

krkrchan, 2023-01-03:
@waliwaliwa 您好，请问您能使用最新的固件嘛，在R2C PLUS 上。我的R2C好像不太行

jjm2473, 2023-01-30:
@krkrchan  试试这里的固件 https://github.com/istoreos/istoreos/issues/566

krkrchan, 2023-01-30:
@jjm2473 经初步尝试，WAN口、LAN口灯都会亮，网络也正常。谢谢！

jjm2473, 2023-01-30:
> @jjm2473 经初步尝试，WAN口、LAN口灯都会亮，网络也正常。谢谢！

重启几次，看看重启以后是不是都正常

jjm2473, 2023-02-02:
试试这个固件有没有问题 https://we.tl/t-N0NuDgYh6i

krkrchan, 2023-02-02:
@jjm2473 您好，今天发的这个固件我测试了，可以正常使用。也重启过，重启后也能使用。

`[   24.709965] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:00] driver [YT8521 Ethernet]`