istoreos/istoreos #372: X86版本网络经常故障

jhding80, 2022-09-09:
X86版本网络经常故障，故障状态就是有线网络断掉，设备需要重启

cnwikee, 2022-09-12:
顶一顶，我的也是，用一段时间后就无网络了， 手动重启WAN口后正常

jannson, 2022-09-22:
那可能是网卡驱动问题，那可能就是缘分未到了。这种问题我们修复会比较慢。可以汇报下你的硬件配置，特别是 CPU，网卡等

cnwikee, 2022-09-22:
系统
主机名
iStoreOS
型号
J3160 AIO
架构
Intel(R) Celeron(R) CPU J3160 @ 1.60GHz
目标平台
x86/64
固件版本
iStoreOS 21.02.3 2022092019 / LuCI istoreos-21.02 branch git-22.258.47264-284140f

***@***.***:~# lspci|grep -i 'eth'
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 07)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 07)

目前的临时解决方案，是写了一个脚本每10分钟ping一下223.5.5.5 连续5次不通判断断网，重启一下网络 /sbin/ifup wan

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

MrShellad, 2022-11-11:
遇到同样问题，经常无法连接网络，重启网口没用，必须重启设备才能重新拨号连接

cnwikee, 2022-11-24:
> 我去，还有这样的使用方法，我们也可以考虑增加这个脚本。。。

2022-11-09 01:00:01 System run times: 219804 sec, The network is normal.
2022-11-09 01:15:00 System run times: 220703 sec, The network is normal.
2022-11-09 01:30:00 System run times: 221603 sec, The network is normal.
2022-11-09 01:45:00 System run times: 222503 sec, The network is normal.
2022-11-09 02:00:00 System run times: 223403 sec, The network is normal.
2022-11-09 02:15:00 System run times: 224304 sec, The network is normal.
2022-11-09 02:30:00 System run times: 225204 sec, The network is normal.
2022-11-09 02:45:00 System run times: 226104 sec, The network is normal.
2022-11-09 03:00:00 System run times: 227004 sec, The network is normal.
2022-11-09 03:15:00 System run times: 227904 sec, The network is normal.
2022-11-09 03:30:00 System run times: 228804 sec, The network is normal.
2022-11-09 03:45:00 System run times: 229704 sec, The network is normal.
2022-11-09 04:00:00 System run times: 230604 sec, The network is normal.
2022-11-09 04:15:00 System run times: 231504 sec, The network is normal.
2022-11-09 04:30:00 System run times: 232404 sec, The network is normal.
2022-11-09 04:45:00 System run times: 233304 sec, The network is normal.
2022-11-09 05:00:00 System run times: 234204 sec, Network Error!!!, restart pppoe.
2022-11-09 05:15:00 System run times: 235104 sec, The network is normal.
2022-11-09 05:30:00 System run times: 236004 sec, The network is normal.
2022-11-09 05:45:00 System run times: 236904 sec, The network is normal.
2022-11-09 06:00:00 System run times: 237804 sec, The network is normal.
2022-11-09 06:15:00 System run times: 238704 sec, The network is normal.
2022-11-09 06:30:00 System run times: 239604 sec, The network is normal.
用脚本后基本无感网络断开过了，无形中呵护你的PPPOE。

enigma84822, 2023-09-21:
请问升级到最新的固件，还有这个问题么？

jhding80, 2023-09-21:
没问题了

| |
Ueinno Technology Co
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2023年09月21日 15:59 |
| 收件人 | istoreos/istoreos ***@***.***> |
| 抄送人 | jhding80 ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [istoreos/istoreos] X86版本网络经常故障 (Issue #372) |

请问升级到最新的固件，还有这个问题么？

enigma84822, 2023-09-21:
感谢，我回家升级一下。

138161, 2024-02-26:
先用电脑接，看有几个网关，我一个光猫出来两个网关
在路由器上ip -6 route看一下自己的默认网关。如果获得的是

default from 2402:f000:x:xxxx::/64 via fe80::xxxx:xxxx:xxxx:xxxx dev eth0 proto static metric 512

这样坑爹的网关，在转发NAT包的时候会有问题，需要把去掉from 2402:f000:x:xxxx::/64这一部分的以后的默认路由添加到路由表中。可以新建一个/etc/hotplug.d/iface/99-ipv6，它的内容是
有效
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
iface=wan6
[ -z "$iface" -o "$INTERFACE" = "$iface" ] || exit 0
ip -6 route add ip -6 route show default|sed -e 's/from [^ ]* //'
logger -t IPv6 "Add IPv6 default route."

这里iface是LuCI里面Network->Interfaces里面看到的名字，一般叫wan6。这个脚本的意思是在wan6起来以后读取默认网关，把带from的内容去掉，再加到系统路由表里。记得
chmod +x /etc/hotplug.d/iface/99-ipv6

有效
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
hwaddr=ifconfig |ip -6 route show default | head -n1 |awk '{print $5}'
route -A inet6 add default gw $hwaddr dev eth0

hunjixin, 2024-04-10:
有解决办法吗，经常挂哦

MrShellad, 2024-04-10:
> 有解决办法吗，经常挂哦

应该是openwrt的驱动版本太老了，

wybinhua, 2024-05-13:
> 请问升级到最新的硬件，还有这个问题吗？ 我的硬件环境： 运营商：联通 AC：倍控J4125 AP：小米6000
> 
> 上周六和今天早上就突然断网，需要重启

你的解决了吗 我的也是这样问题

hunjixin, 2024-05-13:
> > 请问升级到最新的硬件，还有这个问题吗？ 我的硬件环境： 运营商：联通 AC：倍控J4125 AP：小米6000
> > 
> > 上周六和今天早上就突然断网，需要重启
> 
> 你的解决了吗 我的也是这样问题 

更新过一个固件，后来没出现过了，不敢更新了，怕又出事。

wybinhua, 2024-05-13:
我现在使用最新版 也会 你是哪个版本

hunjixin, 2024-05-13:
iStoreOS 22.03.6 2024031514 @wybinhua