istoreos/istoreos #377: istoreos更新9.16版本之后软件源一直报错

lvyremix, 2022-09-16:
[image]
软件源报错 配置里选择不了

lvyremix, 2022-09-16:
[image]

longaga, 2022-09-17:
[image]
无论设置什么源都提示错误。。9.16版本，之前是正常的

NyaNyaXin, 2022-09-17:
遇到了相同的问题

zeusfly, 2022-09-17:
一样的问题，但软件源是能正常切换

paulyinghu, 2022-09-17:
一样的问题，但软件源是能正常切换

molin830, 2022-09-18:
遇到了相同的问题

744129844, 2022-09-18:
遇到了相同的问题

zdj8032, 2022-09-18:
一样的问题，但软件源是能正常切换

lvyremix, 2022-09-19:
好像说明天更新修复 大家等等看吧

jannson, 2022-09-22:
已经修复，你们继续更新版本就好了。

closed 2022-09-22 by jannson

zhouweisong, 2022-10-12:
> 

我已经安装了最新版 还是没有解决 istoreos-21.02.3-2022092414-x86-64-squashfs-combined-efi

zhouweisong, 2022-10-12:
> 已经修复，你们继续更新版本就好了。

我已经安装了最新版 还是没有解决 istoreos-21.02.3-2022092414-x86-64-squashfs-combined-efi

w2816771, 2022-10-20:
我也遇到了同样的问题

Relax-87, 2023-09-13:
@jannson 22.03.5 2023090810 版本 仍然报错 软件源错误 无论切换那个都一样提示

grock304, 2023-09-15:
为啥去年 9 月的问题，我今年 9 月遇到……是一到 9 月就好扯拐吗？

Zhiyuan2024, 2023-09-18:
> 为啥去年 9 月的问题，我今年 9 月遇到……是一到 9 月就好扯拐吗？

一样 今天遇到了

jzsmqss, 2023-12-12:
我用于pppoe拨号，一样的报错，现在是2023年12月12日了，还是修复不了吧？
（源是清华，固件版本iStoreOS 22.03.5 2023112413，内核版本5.10.176