istoreos/istoreos #382: iStore OS 能否集成VPN Server

baobeiboboda, 2022-09-20:
宝哥 可否考虑集成VPN Server 进系统里面

jannson, 2022-09-22:
有考虑，但是还没时间，注意配置好复杂，我也好久没玩过了。这个有官方默认支持的插件，集成就行了。

baobeiboboda, 2022-09-22:
谢谢宝哥 期待 这个 更新 嘿嘿

vinson0231, 2022-10-24:
期待更新，PPTP L2TP

wangchencong66, 2023-01-29:
我也在等更新 PPTP L2TP

meiti789789, 2023-06-08:
坐等加入插件，可以换掉ikuai了

bastonly, 2023-07-23:
网络接口那边是否可以增加支持L2TP的接口呢？ 这样网络就也可以走VPN线路

shwhy1998, 2023-11-15:
其实Istoreos能够集成Ikev2/ipsec 这种就好了，我查阅了蛮多的文章，在Openwrt上貌似没有很好的方案。

wuzhaozhongguo, 2023-11-23:
求VPN server，网上找了各种ipk都装不上，esir的pptp ipk装上后一直显示未运行

merrybip, 2023-12-12:
VPN Server 我安装了几个都不行，需要手动配置，而且接口里面没有VPN，无法正常使用。

yufeiyang2008, 2023-12-14:
> 其实Istoreos能够集成Ikev2/ipsec 这种就好了，我查阅了蛮多的文章，在Openwrt上貌似没有很好的方案。

https://github.com/hwdsl2/docker-ipsec-vpn-server这个了解一下，完美解决

gapacher, 2024-03-24:
> 有考虑，但是还没时间，注意配置好复杂，我也好久没玩过了。这个有官方默认支持的插件，集成就行了。

小宝总， 这个功能可以避免开大量的端口。能非常明显的改善安全性，另外还能共享到家里的科学网络，以及方便不方便开的各种服务。实在是必要， 还请大神能花些时间研究一下。 
PS.我发现unraid已经是一个很成功的商业案例，istoreos可以消防类似的收费模式， 不过的确国内的监管是比较严格的。 还希望你有赚又开心。 祝身体健康，事业顺利！

tttttb, 2024-03-30:
希望集成IPsec VPN 服务器

NinoFocus, 2024-07-10:
我觉得iStoreOS非常好，唯一期待的就是有一个IPSEC VPN服务器，这样我的手机和电脑都可以直接用系统的VPN配置回家了

baobeiboboda, 2024-07-17:
> 有考虑，但是还没时间，注意配置好复杂，我也好久没玩过了。这个有官方默认支持的插件，集成就行了。

现在有一个替代解决方案 就是 安装一个ssserver ，然后配置好 外部端口转发到本机，surge 直接增加一条内网记录，在外面直接输入内网ip 访问

violente21, 2024-08-22:
iStoreOS里怎么又pureVPN？地址192.168.100.199

slb116763814, 2024-09-11:
ipsec这个有没有方法自行安装到istoreOS中？

jazsm, 2024-10-10:
> > 其实Istoreos能够集成Ikev2/ipsec 这种就好了，我查阅了蛮多的文章，在Openwrt上貌似没有很好的方案。
> 
> [https://github.com/hwdsl2/docker-ipsec-vpn-server这个了解一下，完美解决](https://github.com/hwdsl2/docker-ipsec-vpn-server%E8%BF%99%E4%B8%AA%E4%BA%86%E8%A7%A3%E4%B8%80%E4%B8%8B%EF%BC%8C%E5%AE%8C%E7%BE%8E%E8%A7%A3%E5%86%B3)

有，有些固件上集成了这个服务器，很好用，最主要是iPhone自带客户端，不需要下载APP

jazsm, 2024-10-10:
我今天在istoreos固件上安装了openvpn服务器端，但手机要安装客户端APP，还要下载证书，比ipsec麻烦，但速度也很快，连入家中科学取决你上行带宽。

gavin2006001, 2024-10-22:
> > 其实Istoreos能够集成Ikev2/ipsec这样就好了，我查阅了很多的文章，在Openwrt上显然没有很好的方案。
> 
> [https://github.com/hwdsl2/docker-ipsec-vpn-server 这个了解一下，完美解决](https://github.com/hwdsl2/docker-ipsec-vpn-server%E8%BF%99%E4%B8%AA%E4%BA%86%E8%A7%A3%E4%B8%80%E4%B8%8B%EF%BC%8C%E5%AE%8C%E7%BE%8E%E8%A7%A3%E5%86%B3)

我的istore os旁路由安装docker版的ipsec-vpn-server，能安装成功但无论如何都连不上，500和4500的UDP也通过网关转发了，问题到底出在哪？

yufeiyang2008, 2024-10-22:
> 我的istore os旁路由安装docker版的ipsec-vpn-server，能安装成功但无论如何都连不上，500和4500的UDP也通过网关转发了，问题到底出在哪？

之前有有验证过，该方案没有问题，注意下证书安装

gavin2006001, 2024-10-22:
> > > > 其实Istoreos能够集成Ikev2/ipsec这样就好了，我查阅了很多文章，在Openwrt上显然没有很好的方案。
> > > 
> > > 
> > > [https://github.com/hwdsl2/docker-ipsec-vpn-server 这个了解一下，完美解决](https://github.com/hwdsl2/docker-ipsec-vpn-server%E8%BF%99%E4%B8%AA%E4%BA%86%E8%A7%A3%E4%B8%80%E4%B8%8B%EF%BC%8C%E5%AE%8C%E7%BE%8E%E8%A7%A3%E5%86%B3)
> > 
> > 
> > 我的istore os旁路由安装docker版的ipsec-vpn-server，能安装成功但是无论如何都连不上，500和4500的UDP也通过网关转发了，问题到底出在哪里？
> 
> 之前有验证过，该方案没有问题，注意下证书安装

怎么下证书安装，能详细说说吗？ IPsec/L2TP 模式也需要下证书吗？

gavin2006001, 2024-10-23:
> > > > > 其实Istoreos能够集成Ikev2/ipsec这样就好了，我查阅了很多文章，在Openwrt上显然没有很好的方案。
> > > > 
> > > > 
> > > > [https://github.com/hwdsl2/docker-ipsec-vpn-server 这个了解一下，完美解决](https://github.com/hwdsl2/docker-ipsec-vpn-server%E8%BF%99%E4%B8%AA%E4%BA%86%E8%A7%A3%E4%B8%80%E4%B8%8B%EF%BC%8C%E5%AE%8C%E7%BE%8E%E8%A7%A3%E5%86%B3)
> > > 
> > > 
> > > 我的istore os旁路由安装docker版的ipsec-vpn-server，能安装成功但是无论如何都连不上，500和4500的UDP也通过网关转发了，问题到底出在哪里？
> > 
> > 
> > 之前有验证过，该方案没有问题，注意下证书安装
> 
> IPsec/L2TP模式也需要下证书吗？

我的旁路由Docker安装ipsec-vpn-server后使用IKEv2模式通过strongSwan客户端下载证书可以使用，但IPsec/L2TP模式有成功的吗？

apple524, 2025-02-24:
看看今年2025年会有这个软件不